dalathegreat/Battery-Emulator #1: Tesla model 3

jacobbloy, 2023-03-03:
https://openinverter.org/wiki/Tesla_Model_3_Battery

Is there any way to intergrate into the tesla model 3 battery, I know ingen have a controller that has full control of the penthouse.

dalathegreat, 2023-03-03:
@jacobbloy Yes, I am following how two other people are doing their Tesla Model 3 integration (currently working on CAN messages for closing contactors). Once the solution works, we will add it here also. If you are in a hurry, or want to help, join the Patreon Discord server!

jacobbloy, 2023-03-03:
I don’t know if I have the skills, happy to contribute

rand12345, 2023-03-04:
I have also implemented this idea for other battery packs and inverters using the CAN485 board (private) and STM32 (public).

Putting this behind a paywall is not great, I will support actual open source projects but not ones where it's cash for access. Please reconsider your model.

dalathegreat, 2023-03-04:
LOL 😂 @rand12345 , the source for the Tesla WIP code is available here for free: https://gitlab.com/pelle8/gen24 

Like I mentioned, when it works I will put it here also since I intend to use Tesla packs later on. I don't want to put anything untested in this repository.

dalathegreat, 2023-05-03:
@jacobbloy , Tesla Model 3 battery testing has now started. This branch has the code: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/tree/multiple-battery-support for anyone wanting to join in!

dalathegreat, 2023-07-26:
Just wanted to let everyone know that some people are using Tesla Model 3 packs with this solution now. But many are facing issues with getting the HVIL loop satisfied, to get contactor closing. Seems Tesla switched it up massively between Gen3 and Gen4, so we need to investigate this more. Heavy discussion ongoing on the Discord server, let me know if you are testing with a Tesla Model 3 pack and I can invite you!

dalathegreat, 2023-08-20:
@jacobbloy Tesla Model 3 packs now in use, tagged the latest version! https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/releases/tag/v2.5.0

Closing this issue as completed!

closed 2023-08-20 by dalathegreat