dalathegreat/Battery-Emulator #7: Chademo

MArul1989, 2023-03-27:
Evening all,

Is is possible to leave the battery in the car and use the LilyGo, some electronics to simulate the Chademo signals the correct solar inverter ( 50Hz, single phase <5kW, with the RS485 for me) and a Chademo cable.  I know these can be bought, but not that available in the UK and pricey.

https://www.researchgate.net/figure/Flow-diagram-of-the-CHAdeMO-chagrining-protocol_fig4_319162700

Insulation test an issue!

https://www.thingiverse.com/thing:121581

Would have to make waterproof, more sturdy etc.

Thanks for your attention.

dalathegreat, 2023-04-02:
To do this safely, you will need a proper Chademo cable. These cost well over 1300€, for the same price of a Chademo cable you can get a 24kWh battery in some parts of the world, so it is not really worth it in my opinion.

Probably someone can make something work, but this repository is all about re-using used EV packs for stationary storage, and let's keep it that way. Closing this!

closed 2023-04-02 by dalathegreat

MArul1989, 2023-04-02:
Thank you for responding.  I'll keep in mind that  future comments are  relevant to the repository.