dalathegreat/Battery-Emulator #9: Voltage range on LilyGo

roelof4, 2023-04-24:
Nice job!

I like the LilyGo TCAN board as it seems to have everything needed apart from wired ethernet.

Question: Looking at the schematic it should be possible to power this board unto 16V (the absolute max for the on board regulator). The FET in line is capable with handling even more. So why the note that one shouldn't go beyond 12.3V?

dalathegreat, 2023-04-24:
2 persons have messaged me stating that their LilyGo boards went out with 13V input voltage. And on their website they state 5-12V input voltage max also! So best not to risk it, also possible to use a 9V input!

roelof4, 2023-04-24:
Strange - there is even a diode in the path which would drop another 0.7V. Ah well perhaps these regulators are sub spec.

closed 2023-04-24 by dalathegreat