dalathegreat/Battery-Emulator #10: Why not using relay for A/B/C/D switch control ?

X-Ryl669, 2023-05-26:
I wonder why you're using a manual switch for A/B/C/D instead of a (cheap) 4 relay board to control them for the precharge/charge sequence?

You can't respect the tight timing with manual switch (hard to switch on and off in less than 0.3s).

dalathegreat, 2023-06-09:
Battery agnostic precharge/contactor closing has been added to the software. https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/blob/main/Software/Software.ino#L206

Still needs to be tested and documented!

bobbo3, 2023-08-15:
We did some testing and on the battery we used the bit for LB_MainRelayOn_flag is always 0.
I changed the code temporarily so batteryAllowsContactorClosing = 1; and the precharge worked fine. 
This was on an 30kWh battery. 
Here is an screenshot on the relevant byte (3) from a log file.
[image]
I am not very familiar with Leaf battery and will appreciate if anyone has any ideas on what to test.

dalathegreat, 2023-08-15:
Thanks for testing @bobbo3 , you beat me to it! LB_MainRelayOn_flag wont be set to 1 unless both HV connectors are seated. Do you have both the big and the small seated?

bobbo3, 2023-08-15:
Thank you! I think we shorted both interlocks - the drive one and the heater one... but it is possible that we may have missed something. I will double check.