dalathegreat/Battery-Emulator #11: Use whole Leaf as a battery for Hybrid Solar inverter?

fototakas, 2023-05-27:
Hello Dala,
Thanks for sharing another nice project.
What do you think is it possible to use same approach and connect Leaf using CHAdeMO to Hybrid inverter as HV battery? in theory we would be able to charge Leaf with DC from panels, and use car for emergency backup. I must admit, that most expensive part will be the CHAdeMO plug :)

dalathegreat, 2023-05-28:
Hi @fototakas 

See this reply here: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/issues/7#issuecomment-1493375939

Closing this as duplicate

closed 2023-05-28 by dalathegreat

fototakas, 2023-05-29:
@dalathegreat Found Chademo plug with cable for the good price, still 2 available:
https://www.ebay.com/itm/235021221143