dalathegreat/Battery-Emulator #12: Compatibility with Other Inverters

candycab, 2023-06-01:
Hi.
Acquiring this Fronius Gen24 is proving to be difficult in my area.
Has anyone tested this project with other types of inverters? Are there any alternative inverters that would be compatible? Any suggestions or shared experiences would be highly appreciated.

RobDuthie, 2023-06-01:
Are you trying to do a microgrid with it? As he lateset firmware for primos etc had lots of issues with the microgrid tied to a Victron, Fimrer worked Ok, Huawei is OK, and Sungrow is OK.

abraha2d, 2023-06-04:
I'm looking into adapting this to work with SolarEdge inverters (and Tesla batteries, but #1 already exists for that), however it might be some time before I get around to actually poking at it.

dalathegreat, 2023-06-09:
@candycab Sungrow inverter compability is being added right now via emulating a BYD HVS via CAN! This opens up for quite a lot of compatible inverters, see this list: https://www.bydbatterybox.com/uploads/downloads/BYD%20Battery-Box%20Premium%20HVS&HVM%20Compatible%20Inverter%20List-V2.10-63d76b3b31b89.pdf

dalathegreat, 2023-07-26:
Closing this ticket, inverter compatibility has been increased significantly in the last months:
[image]

See this link for info: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/wiki#supported-inverters-list

closed 2023-07-26 by dalathegreat