dalathegreat/Battery-Emulator #13: Q: house battery

boelle, 2023-06-01:
i plan to use the battery as a house battery without solar, but charge when electricity is cheap and then use the battery when its expensive

what chargers can be used? i tried to look for a 3phase charger that ouputs 400v but they are hard to find, most are 48V or below

inverters.... are there any that accept the full pack voltage?

or do i have to reconfigure the battery for a lower voltage?

jymbob, 2023-06-01:
You need a BYD HVS compatible inverter. Don't know where you are in the world, but a UK supplier has this handy chart to start with https://www.bydbatterybox.com/uploads/downloads/BYD%20Battery-Box%20Premium%20HVS&HVM%20Compatible%20Inverter%20List-V2.10-63d76b3b31b89.pdf

boelle, 2023-06-02:
> Don't know where you are in the world,

DK so on the other side of the north sea

my thoughts and they might well be total nuts, but i had the thought of charge the battery when its cheap, how long will it take? my electricity company uses 1 hour slot so if it can be done in 1 hour that would be the best, if not i need to program a pi to gather prices from nordpool and pick the cheapest 2 or 3 hours

next would be to have enough capacity to run for a full 24 hours, in december the worst day was just shy of 80kwh 

what is the best or cheapest way to tackle it?

skaggetse, 2023-06-03:
> DK so on the other side of the north sea
>
>
>
> my thoughts and they might well be total nuts, but i had the thought of charge the battery when its cheap, how long will it take? my electricity company [...]

Solar Panels is the way to go.

I don't know the specifics of the danish electric fees, but it might be a really bad ROI of such an investment without solar panels.

In sweden, my fee excluding the actual electricity price is 1SEK - plus an effect fee. So the difference between the highest and lowest spot price must be at least 1 SEK including taxes, only to go flat even. That does happen, but it is not as common as people seem to think.

And you will not be able to find a charger for consumer price to be able to charge a full battery in an hour. Not even the 24kWh one.

And this does not take in count the degrading of the battery. 

In my oppinion, all the home-battery-products that have emerged these last years are all pure scam, theres no way to get an ROI off of them to actually make them worth it.

boelle, 2023-06-03:
what is the fastest i can charge then?

and yes, that is why i look in to mixing up my own product, i can only blame myself if something is not right

skaggetse, 2023-06-03:
> what is the fastest i can charge then? assuming that i use all 3 phases at 16 amps each
> 
> 
> 
> and yes, that is why i look in to mixing up my own product, i can only blame myself if something is not right

3phase 16a is 11kW.

boelle, 2023-06-03:
so yeah.... that will be 8 hours.... 

and out of those maybe 3 will be cheap

boelle, 2023-06-03:
i think i  found an battery pack at 40kw that cost arround 66000dkk, i would need 2 of them to cover a bad winter day, my house is with heat pump and there are no other options as i live 400 meter from the coast, that is why i have so high an usage in the winter

the more windmill we get the better as it will have a surplus in the winter and that gives low prices, is the summer its the solar farms that generate enough that we each week get days where the prices goes negative, we do not get paid to use but it means free electricity

i tried to look at january and december last year and figured i could have saved about 1000 DKK a month if i could fit my usage within the cheapest hour, so given that i would need the cheapt 8 hours out of 24 i would say that i could save 500 DKK instead

22 years just to recover the cost of the battery, and yeah they will not last that long, and they are not new to begin with

so starts to look like a very bad idea, and that is even for something i build up myself

dalathegreat, 2023-07-03:
I am closing this issue. Questions like this is more suitable for discussion forums, like Secondlifestorage.com

For currently supported batteries/inverters, see the wiki: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/wiki

closed 2023-07-03 by dalathegreat