dalathegreat/Battery-Emulator #15: huawei inverter

MotorBlock, 2023-06-24:
Looking for software to add leaf battery to a huawei sun2000 hybrid inverter

dalathegreat, 2023-07-25:
It is coming later on when I have time!

For now I added the following inverters to the supported list:
[image]

dalathegreat, 2023-08-20:
@MotorBlock and @Turbobussi Huawei inverter integration now started! Added it to the wiki!

spelzl, 2023-08-20:
Great news!
Please shout out if you need more logs...

DanieleCar, 2023-08-22:
Hi great news , huawei inverter divide in 2 family : the old sun2000L that accept only the LG resu battery ( High voltage) and the sun2000 (newer version) that support low voltage huawei luna battery .
on wich one are you working? if i found a battery at a decent prize i will be a beta tester

dalathegreat, 2023-08-22:
@DanieleCar 
We have logs from
"Sun2000 10-KTL-M1" coupled with a "Luna2000-5-S0 5kWh"

DanieleCar, 2023-08-22:
my error , also luna battery are High voltage 

In this document look like that both type of inverter can work with lg resu 
[SUN2000L-(2KTL-5KTL) and SUN2000-(2KTL-6KTL)-L1 Battery and Smart Power Sensor Quick Guide.pdf](https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/files/12410489/SUN2000L-.2KTL-5KTL.and.SUN2000-.2KTL-6KTL.-L1.Battery.and.Smart.Power.Sensor.Quick.Guide.pdf)

spelzl, 2023-08-22:
The -L models are single-phase whilst the -M models are 3-phase.

dalathegreat, 2023-08-22:
We don't have the LG RESU logs, so the first DIY packs here are gonna have to be 800VDC

Also if you use LG RESU, it was a note in the manual "When the battery (LG) is configured, the input voltage of
the PV strings can not exceed 495 V.",, this is a problem for many, my Huawei PV strings go to 800VDC

I added this info to the wiki, thanks @DanieleCar for the link

robbedenys, 2023-08-23:
The monophase SUN2000-L inverters can also work with the luna batteries (350 - 560V), so if you would have the logs of such a combination it is not necessary to put two packs in series to get 800V? The 495V input limit is only if you use the LG RESU battery.

spelzl, 2023-08-23:
@robbedenys true, but I have a three-phase M1 inverter.

robbedenys, 2023-08-23:
What is needed to obtain those logs? My brother has a SUN2000-3KTL-L1 - LUNA 5kWh combination, so I could try to get logs from his installation :-)

DanieleCar, 2023-08-23:
as per huawei : https://forum.huawei.com/enterprise/en/huawei-statement-letter-sun2000l-sun2000-3-10ktl-m0-pv-inverter-and-luna-2000-batteries/thread/667248938798956544-667213868771979264
sun2000L only lg resu 7h or 10h (400v) ( older version of inverter before huawei startet to product battery)
sun2000 single phase (L1):Luna2000 (circa 400 per single phaseapplication) and maybe lg resu  7h or 10h (400v)
sun2000 3 phase (m0/m1): only Luna2000 (circa 600 per 3 phase application)

In my case (sun2000L) the compatibility with more moder luna battery is only a problem of comunication protocol.

nigschs, 2023-08-24:
> @DanieleCar We have logs from "Sun2000 10-KTL-M1" coupled with a "Luna2000-5-S0 5kWh"

Is it possible to share these logs?
Is there an estimate of how long it will take to integrate the Luna2000?

dalathegreat, 2023-08-28:
@nigschs Here is a log of (Sun2000 10-KTL-M1 + Luna2000-5-S0 5kWh):
https://we.tl/t-HNMUSylvey

I have started the decoding on my Discord server, we will commit this to a feature branch soon!

robbedenys, 2023-08-31:
I have another question about the huawei inverter/battery combination. 

I found these specifications of the huawei monophase inverters/batteries:

Nominal voltage (single phase system) 450 V
Operating voltage range (single phase system) 350 – 560 V

When one would connect for example a Tesla model 3 pack (Nominal Voltage : 355V, Fully Charged : 403V, Fully Discharged: 288V), does that mean that only 47% of the battery's capacity can be used?

nigschs, 2023-09-27:
@dalathegreat Have you already succesfully connected a Leaf battery to a Huawei Sun2000 Inverter? As far as I know the battery connection is directly connected to the DC-Link of the inverter which is dependent on the PV input voltage, but at least 600V. At least this is what I measured on my inverter. Isn`t there a need of an additional DC/DC converter to fit the battery voltage to the voltage of the DC-Link of the inverter?

kalectro, 2023-10-13:
let's support his Patreon! I was looking for the ideal solution for a friend who lives a little off-grid in Portugal and I think this is it. I want this :)

Jgfyyvzss, 2023-12-03:
Eagerly awaiting this addition.

I am struggling to understand the 400V/800V problem.
The Huawei Luna2000 specs indicate all batteries are rated 385V, Range 350V - 560V.  I can't find any Huawei reference to 800V LUNA2000 batteries, they do not appear to exist.

I also can't see that connecting an 800V battery to a single phase SUN2000 is going to end well!

Can someone clarify for me?

Lukestaz, 2023-12-15:
Hi folks, Sun2000L 5KTL owner. Keen to assist if feasible to add support. Thanks 

Attached doc quite comprehensive. As already noted limited to KG RESU and max 495V strings.

LG RESU7H (Type- R) and LG RESU10H (Type-R) 

[Sun2000L battery installation .pdf](https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/files/13688476/Sun2000L.battery.installation.pdf)

Artilect01, 2023-12-18:
Modbus onterface definition for Huawei inverters, power meters and Powerwall. I found this on internet a wile ago. If it helps to get the suport for Huawei inverters. 

BR
S

[SUN2000_Modbus-interface-definitions.pdf](https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/files/13703223/SUN2000_Modbus-interface-definitions.pdf)

DanieleCar, 2023-12-19:
Hi , i just bought a lg Resu 10h , if you want a can try to grab some logs 

My setup is : Huawei Sun2000L - Lg Resu 10H - power meter

MayBankCH, 2023-12-26:
> Eagerly awaiting this addition.
>
> I am struggling to understand the 400V/800V problem. The Huawei Luna2000 specs indicate all batteries are rated 385V, Range 350V - 560V. I can't find any Huawei [...]

DC/DC Converter
[image]

[image]

michak89, 2024-01-21:
Whatever you guys are doing here is groundbreaking, so far no one has connected a battery other than original to this huawei inverter. I unfortunately own 8Ktl (no battery), happy to help with any tests.

emkurten, 2024-01-25:
I have a sun2000 and leaf battery ready and waiting to be hooked up. Would be happy to help with any testing. Looking forward to this to becoming a reality!

spelzl, 2024-01-25:
> I have a sun2000 and leaf battery ready and waiting to be hooked up. Would be happy to help with any testing. Looking forward to this to becoming a reality!

@emkurten Please join Dala's discord channel at (https://discord.gg/KfqUfzaE) and link to the Huawei section.

VeikkoPeikko, 2024-02-02:
I am really exited about SUN2000 integration for high voltage applications. I have two 10KTL-M1 systems running and I have been planning for battery storage for a while. In my calculation reasonable break even level for battery storage investment is around 100-150€/kwh and battery emulation and EV pack is hitting the mark.

Are there any ideas yet, which battery configuration would be ideal for this?

Any idea how to setup communication in configuration where two batteries are connected in series? I did not find any resources or documentation how this would be done in practice. In long term some balancing between two packs is probably needed. 

Hyundai Ioniq 5 batteries seems to be the best option at this moment. As I understood, Ioniq 5 77,4khw pack contains 32 modules (6s2p) and voltage range would be about nominal 697,5V max 806,4V and min 576V. These are quite readily available and price range is quite reasonable. This would sound very feasible solution as single battery once we get communication sorted out. Actually my father (owner of the other SUN2000 system) drives Ioniq 5. Maybe I should get to logging with his car.

dalathegreat, 2024-02-07:
Batterysystem (LUNA2000) is a Battery including DC/DC. So this system is more like a current source. Voltage is given by the DC-Link of the inverter. This DC-Link is depending on the PV-Voltage. If PV-Voltage is less than 600V, the DC Link is at around 600V. For PV-Systems with a String voltage of >600V the DC-Link Voltage is higher. **This means there is no way to directly connect a battery to the SUN2000 Inverter**. There is always a need of an additional DC/DC stage

Copied info from Simon, seems like the Huawei is a dead-end when it comes to cost effective battery storage :(

spelzl, 2024-02-07:
There might still be interest for an EV battery capacity extension for those who already have a Huawei Luna2000-C0 BMS installed or willing to buy one.

WiebKastanje, 2024-02-09:
@spelzl , yes I'm very interested in battery capacity extension.
Huawei is definitely not a dead-end when it comes to cost effective battery storage.
My setup: SUN2000-8KLT-M1 + Luna2000-C0  (Battery Management System) + Luna2000-5kW-E0 (Battery)

[image]

cikamartin, 2024-04-30:
Newbie here, I have SUN 5TKL i 10TKL and nissan env200 24kWh battery on hand.

Since 5kW inverter works with voltage around 400V, I was wondering why wouldnt I directrly connect battery to second string that is not used?

Inverter could not pull more then 5KW, only problem would be charging the battery.

What am I missing :)

spelzl, 2024-05-07:
Hi @cikamartin 
I don't think that you are missing anything. In fact, you are probably right. It would require someone to read, analyse and interpret the battery-internal communication between the battery and BMS modules.
If you can take on this challenge please join Dala's discord channel and become an active member of the community.
You are welcome.

cikamartin, 2024-05-10:
> Hi @cikamartin I don't think that you are missing anything. In fact, you are probably right. It would require someone to read, analyse and interpret the battery-internal communication between the [...]

I am almost done with discharging part, planing on using liligo or OBDII for comunicating with battery and definetly undervoltage protection (and overvoltage, once I can charge it :) )

I hope that I will, in next 10 days, be able to test if it will work (at night)

Will join discord as soon as I discover how :)

spelzl, 2024-05-10:
[image]

Download the app and search for Dala's EV Discord

[image]

spelzl, 2024-05-22:
Hi @cikamartin ,
can you please share more details about your project and how it connects the EV battery to the Luna2000?
What EV battery make and model are you using?
How far have you progressed the charge - discharge?
Have you meanwhile joined Dala's discord group? Please refer to the huawei-luna2000-reverse-engineering sub-section
https://discord.com/channels/775038294655762433/1064668460363296899

cikamartin, 2024-05-23:
sorry, didnt have much time for wiring. im checking is it cheaper to replace huavei inverter :)

edit:
just realised that OBcharger is acctualy called voltage converter, so I found one on ebay. maybe ill go that way...

It is from a leaf

peternooy, 2024-08-31:
> @nigschs Here is a log of (Sun2000 10-KTL-M1 + Luna2000-5-S0 5kWh): https://we.tl/t-HNMUSylvey
> 
> I have started the decoding on my Discord server, we will commit this to a feature branch soon!

Can you share the logs again as the link shown here is expired?

Marcmk1, 2024-09-23:
I've got a SUN2000-6KTL-M1 3-phase.
I will follow you guys for status referse Engineering of this inventor.

It would be awesome if I can connect a car battery to my system!

dalathegreat, 2025-10-07:
No progress for 2½ years. See the Wiki. https://github.com/dalathegreat/Battery-Emulator/wiki/Inverter:-Huawei-SUN2000

Closing this issue, dont think we'll ever see the DC/DC development needed to boost to 900V