dalathegreat/Battery-Emulator #16: Battery compability list

bruvv, 2023-06-27:
In the readme you write: "You will also need a complete EV battery. See the battery compability list on which are supported." where is this list?

AlexJamesHaines, 2023-06-28:
This is a valid question and I haven't spotted it on quick inspection of the code. I'd propose that early 30kwh leafs are a good choice but it would be nice to have the list somewhere 👍

bruvv, 2023-06-29:
@AlexJamesHaines I think tesla battery is also supported: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/blob/main/Images/Tesla_X098.png

dalathegreat, 2023-07-01:
@bruvv and @AlexJamesHaines , see the Wiki section: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/wiki#supported-batteries-list

AlexJamesHaines, 2023-07-01:
Not sure how I missed that. Thanks 👍

closed 2023-07-02 by bruvv