dalathegreat/Battery-Emulator #18: Renault Zoe/Kangoo support added, but not tested

dalathegreat, 2023-07-03:
I recently added initial support for Renault batteries, but this needs to be tested extensively. Please contact me if someone here is planning to use these batteries!

https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/commit/caeb52e4aa7e904f3005e71d55063f730ff45956

rand12345, 2023-07-12:
[image]

Why didn’t you contact the me? You already know that have the complete Kangoo and Zoe protocols tested, the complete Can485 hardware setup with dual can, contactor control etc., and a fully tested STM32F105 implementation for all the protocols (literally 1:1) as my repo. 

Not cool.

dalathegreat, 2023-07-13:
@rand12345 
I used this Renault Zoe CAN matrix spreadsheet: https://docs.google.com/spreadsheets/d/1Qnk-yzzcPiMArO-QDzO4a8ptAS2Sa4HhVu441zBzlpM/edit?pli=1#gid=0

I have also credited your "https://github.com/rand12345/solax_can_bus" repository under the Dependencies: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24#dependencies- Massive thanks for the reverse engineering!

Where is that screenshot from? If you want to share more, drop a link!

rand12345, 2023-07-13:
Don't be coy, this isn't about one Solax protocol and you gave it away in your latest YT video. You were aware of my STM32 repo back in March 2023 and the skeleton of your repo now matches the battery protocols listed in my (then public) repo 1:1. 

https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/issues/1#issuecomment-1454756240

That public STM32 repo was taken private short of mothballing the STM32F105 can filter board. The hardware failure rate is too high for safety critical systems and there are better (and cheaper) solutions out there. Same for the TWAI interface which you have attempted to compensate for with CRC (useless for other battery protocols).

This is what happens when actual development is held behind a **pay feature**, you lose the support of devs who have these batteries on-site for reverse engineering. Good luck with YouTube bucks.

dalathegreat, 2023-07-13:
@rand12345 
Here is an invite to my Patreon Discord server: https://discord.gg/mEUe2kgf
This link is valid for 7 days. 

Please join in on the discussion, I am having a very hard time understanding what you are upset about. Maybe you can join and explain what the issues are?

Headline81, 2024-04-25:
I have a Zoe 41kwh pack. Already running a Leaf-pack. What do we need to test it out? I´m not a SW guy so i can not du any code.

Zoe22KW, 2024-05-04:
Hi everyone, i try with Zoe 22KW but without success. Error message: Check wires-CAN fault.

Can someone help me ?

Thanks a lot. Sorry about my english.

dalathegreat, 2024-07-01:
@Headline81 and @Zoe22KW , I added new info about Zoe, I have split the code into Gen1 22kW and Gen2 41kWh 3 weeks ago:

https://github.com/dalathegreat/Battery-Emulator/pull/343

You can try the latest version of the emulator, and see how it works now! Feel free to report back :) I will close this issue, and we can open separate issues for each generation incase there are any problems!

closed 2024-07-01 by dalathegreat