dalathegreat/Battery-Emulator #24: Leaf CRC improvement

AlecDavisNZ, 2023-07-22:
The CRC calcualtion returns 0 if you include the actual CRC which is normally the last byte.

```
uint8_t calc_crc8( uint8_t *data, uint8_t len ) {
	uint8_t crc = 0;
	uint8_t i;
	
	for ( i = 0; i < len; i++) {
		crc = crctable[ (crc ^ data[i]) ];
	}
	return crc;
}

```
The use case for checking a received frame of 8 bytes, could look like;
```
  case 0x1DB:
    if ( calc_crc(&rx_frame.data.u8[0], 8) ) {
      CANerror++;
      break; //Message content malformed, abort reading data from it
    }

```
instead of
```
  case 0x1DB:
    if(is_message_corrupt(rx_frame))
    {
      CANerror++;
      break; //Message content malformed, abort reading data from it
    }

```
**BONUS**
The use case for transmitting a frame thats requires a CRC, Like 0x50C which has a DLC of 6 to keep the SOC alive

`LEAF_50C.data.u8[5] = calc_crc8( &LEAF_50C.data.u8[0], 5);  //crc of bytes[ 0-4 ]
`

Note: This code is tested, but not ion the arduino environment, it's in a Zilog chip :)
I've converted names etc to match your's hopefully I got them right.

dalathegreat, 2023-07-26:
Thanks @AlecDavisNZ , some solid improvements here! I will take this into use on next software version!

dalathegreat, 2024-04-25:
Closing as wont do, I experimented with this type of functionality on the BMW i3 (having same CRC function for checking and signing, but doing a quick `if(is_message_corrupt(rx_frame))` makes the code so much more newcomer friendly!

closed 2024-04-25 by dalathegreat