dalathegreat/Battery-Emulator #28: Goodwe Inverter canbus issue

JHendrikz, 2023-07-26:
Hi all,

Just looking into the compatibility of emulating the BYD HVS battery with a GoodWe GW6000-EH via Canbus. Everything is hooked up as expected between the inverter and the LilyGo, simulating battery data until I can prove comms with the inverter.

I've scoped the canbus signal and have the following Goodwe inverter output shown below

0x151      00 47 4F 4F 44 57 45 00
0x191      04 04 00 17 00 01 04 00

[image]

I'm assuming the 0x151 signal should have a 0x01 in byte[0] but this doesn't seem to be the case. Has anyone come across this before? A box of beers for anyone who can steer me in the right direction!

Cheers

dalathegreat, 2023-07-26:
@JHendrikz , did you enable the "#define CAN_BYD" in the Software.ino? After doing this, it will send the startup CAN data once, and it will re-send it incase it sees 0x151 with 0x01 in frame1. 

Also, did you enable battery in the inverter settings? Not sure if the Goodwe inverter needs to be configured before starting to act on the CAN messages.

JHendrikz, 2023-07-27:
Hi Dala, thanks for your feedback and yes running the CAN_BYD enabled but still have a battery comms error on the inverter setup app.

Looks like it might be something at the inverter side so I will do some more investigation and let you know, I will note that the Goodwe seems to be quite finicky/glitchy with its initial configuration setup.

One question I had was that currently I have isolated the battery output (to make sure the comms are working). I imagine the battery comms should still work with the battery isolated from the inverter? It would be somewhat of a safety risk if this wasn't the case. 

Will keep you posted on what I find.

dalathegreat, 2023-07-27:
@JHendrikz try the latest released v2.1.0 version, it now makes the LED on the board blue incase the CAN wiring is incorrect! https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/releases/tag/v2.1.0

In your log I don't see any CAN messages sent from the LilyGo board, so I suspect you might have wiring issues

JHendrikz, 2023-07-30:
Hi Dala,

The CAN wiring was correct, I just sent hadn't sent the startup CAN initialization data. The inverter also required the battery to be connected to the inverter before the comms would show up as functional. Found two bugs with the BYD code as shown below:

```
BYD_110.data.u8[4] = (max_target_charge_power >> 8);
BYD_110.data.u8[5] = (max_target_charge_power & 0x00FF);
```

Should be max_target_discharge_current:
```
BYD_110.data[4] = (max_target_discharge_current >> 8);
BYD_110.data[5] = (max_target_discharge_current & 0x00FF);
```
and
```
BYD_110.data.u8[6] = (max_target_discharge_power >> 8);
BYD_110.data.u8[7] = (max_target_discharge_power & 0x00FF);
```

Should be max_target_charge_current
```
BYD_110.data[6] = (max_target_charge_current >> 8);
BYD_110.data[7] = (max_target_charge_current & 0x00FF);
```

```
max_target_charge_current = (max_target_charge_power*10)/LB_Total_Voltage;
max_target_discharge_current = (max_target_discharge_power*10)/LB_Total_Voltage;
```

Other than that the GW6000-EH appears to be operating as expected, will monitor for any issues/bugs going forward. Thanks once again @dalathegreat for the excellent bit of software.

dalathegreat, 2023-07-30:
Thanks @JHendrikz , merged PR #38 ! Please let me know if you still have any feedback 😁

JHendrikz, 2023-10-10:
Hi @GubbJVL, happy to help with your GoodWe installation. 

Have been running mine on and off for about 2 months now. Currently chasing a bug on the SOC% not displaying correctly on the SolarGo app but other that it is working pretty well.

I'm running a version of Dala's code, however I have rewritten in a FreeRTOS environment so not exactly the same. Let me know how far you are with your installation or what you need particular help with and we can go from there.

JHendrikz, 2023-10-12:
Down in New Zealand and not on discord at the moment. The setup with goodwe was relatively straight forward and easy to source here in NZ. 

1. Follow the goodwe inverter installation manual up to the point when you plug the comms and battery connector terminals in
2. Unsolder the canbus terminating resistor on the Lilygo board
3. Connect both the BMS canbus and inverter canbus to the board
4. Double check your code to make sure the contactors will isolate under the required fault conditions
5. Plug in the battery connector (make sure the terminals are not live)
6. Double double check all your wiring
8. Switch your battery breaker on
9. Enable the contactors and the inverter will configure and connect if all has gone well.
10. Configure the inverter with the battery connected (this is where the goodwe software can be a little glitchy)

That's more or less the approximate process, the inverter setup is pretty glitchy so once you get to that point I can provide a bit more step-by-step info/videos.

Hendersk, 2023-10-14:
Hi @JHendrikz , I am also in NZ (Hamilton) and was hoping to reach out and see if you hade some time to help. I have gone as far as I can with my knowledge but still haven't got this working. I am using a goodie 10kw ET series, Nissan Leaf 24 battery and the coding and lilly go as recommended on Dalas threads. I have followed his guides very closely. 

I can run the battery through the goodwe in manual define mode (no BMS) but this is obviously a bit risky, I only make the point to confirm battery is energised fine and supplying voltage. The lily glows green when connected to the leaf battery, goes blue if I unplug it, so I think the can from the battery is working. I have tried both the BYD modbus and the CAN options in Dallas setup file but both options present with what appears to be communications errors CRC check error, packet length etc. And when trying to configure the battery to BYD HVM or HVS in the software the inverter informs communications error. Any ideas? Can get in direct contact with you via email? 

Thanks
Regards
Kane

ssamoy-poulpharm, 2023-10-16:
> 9\. Configure the inverter with the battery connected (this is where the goodwe software can be a little glitchy)
> 
> That's more or less the approximate process, the inverter setup is pretty glitchy so once you get to that point I can provide a bit more step-by-step info/videos.

Hi J

> Hi @JHendrikz , I am also in NZ (Hamilton) and was hoping to reach out and see if you hade some time to help. I have gone as far as I can with my knowledge but still haven't got this working. I am using a goodie 10kw ET series, Nissan Leaf 24 battery and the coding and lilly go as recommended on Dalas threads. I have followed his guides very closely.
> 
> I can run the battery through the goodwe in manual define mode (no BMS) but this is obviously a bit risky, I only make the point to confirm battery is energised fine and supplying voltage. The lily glows green when connected to the leaf battery, goes blue if I unplug it, so I think the can from the battery is working. I have tried both the BYD modbus and the CAN options in Dallas setup file but both options present with what appears to be communications errors CRC check error, packet length etc. And when trying to configure the battery to BYD HVM or HVS in the software the inverter informs communications error. Any ideas? Can get in direct contact with you via email?
> 
> Update 17/10/23 - @JHendrikz , false alarm mate, I blundered on and now have a working system. Its working quite well actually with Goodwe GW10KL-ET. Ultimately it was an amateur mistake that was holding me up, Thanks to you and @dalathegreat too for the guidance. Cool stuff!
> 
> Thanks Regards Kane

Hey Kane,

Do you want share your amateur mistake with another amateur? :-)
I just hooked up a Tesla model 3 battery to my Goodwe GW10KL-ET but I can't get the communication with the inverter up.
Contactors are closing fine, but BMS communication has a status 'Abnormal'. 

Kind Regards,
Steven

Hendersk, 2023-10-17:
Hi @ssamoy-poulpharm ,

Yeah sure thing. A number of contributing factors actually as follows;

- As per Page 4 of the manual, item 5 is shown as having a cover or a dual gland plate. Well my inverter had two ports labelled as BMS! The BMS lan port I was initially using was n the same land plate as the smart meter connection, after some hair pulling I realise the inverter had another port labelled BMS and this one was next to the DRED, so took that gland plate off and found another LAN port which actually worked!
- My Lillygo go was not dual CAN out, it had CAN on the top and 485 on the bottom and so I had interpreted from the goodie manual page 16 showing the pin outs including 485 to mean that I could use 485 on this inverter as long as connecting to the right pins. I tried everything to get 485 working and couldn't, have learnt its not possible with this inverter, you need to use CAN
- Need to remove CANBUS terminating resistor as per other guidance on this site and connect both the inverter and the battery on the same CAN terminals on the Lillygo.
- After sorting the above, it worked and I was able to configure BYD HVM (or HVS) battery in the Solar and all other settings seem to be working fine. 
I hope this helps!
Regards

Benik3, 2024-06-10:
Hi.
I tried to simulate BYD battery with GW30k-ET inverter.  
The communication looks working, but for some reason, the inverter ignores current limits for charging and discharging.  
Even that in the app it shows correct values (same as sent over CAN).  
Did you noticed any problem like this?  
Thanks

dalathegreat, 2024-06-10:
Hi @Benik3 , I suggest you make a new issue for this, and put lots of screenshots attached to it!

StarkJohan, 2024-06-10:
Also check the battery data in the goodwe solargo app (or semsportal but it hasn't got as nice view of the battery data as the solargo app)

Benik3, 2024-06-11:
Thanks for replies.  
I have the protocol implemented in own BMS.  
In SolarGo App everything looks good (even the limits are shown correctly).  
Maybe it's some problem with just the 30kW version of inverter....

I will try to dig more, if I will find something, I will report back :)

dalathegreat, 2024-06-11:
Ah, if you are not using our code, then we cannot really help that much with troubleshooting. Good luck!

StarkJohan, 2024-08-14:
Yes, during the setup in the GoodWe app (for selecting battery type etc), the battery DC voltage needs to be on and the emulator should be restarted a few seconds before the battery selection is done in the app.
That way I've had the best outcome when configuring the Goodwe ET.

StarkJohan, 2024-08-14:
That would need to be addressed with the code for that specific battery and I haven't got any experience with other batteries than the Leaf pack I use.