dalathegreat/Battery-Emulator #42: Byd battery emulator for Sungrow hybrid inverter.

EnergibatteriMax, 2023-08-11:
Hi, I have issues connecting 485 communication to from liligo to sungrow hybrid inverer. 
Has anyone experience from this?
Modbus Framesize errors when monitoring the communication..

anton4, 2023-08-16:
SH5.0RT/SH6.0RT/SH8.0RT/SH10RT models support Li-ion battery from SUNGROW, LG Chem, BYD and Pylontech.
Sungrow, BYD and Pylontech use CAN for communications and LG Chem uses RS485 (modbus)

dalathegreat, 2023-08-17:
@EnergibatteriMax 
Yes you put both battery and inverter to the same CAN port. Also one resistor needs to be removed from the board: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/blob/main/Images/RemoveThisWhenUsingCANinverter.png

We need to create a dedicated wikipage for the Sungrow inverter, many are already using these inverters successfully, but nothing has been written down!

dalathegreat, 2023-08-17:
Sungrow wiki page created: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/wiki/Sungrow-inverters

dalathegreat, 2023-08-23:
I am closing this issue, if you find something you want to improve the wiki with, feel free! 😄

closed 2023-08-23 by dalathegreat