dalathegreat/Battery-Emulator #51: Adding support for other batteries

torsteinelv, 2023-08-20:
What if we want to contribute to add support for other batteries? How can we do that?

dalathegreat, 2023-08-20:
@torsteinelv 

- Get a CAN log of the battery running in a vehicle
- Take pictures of HV and LV connectors. Figure out the pinout with manuals. Add to wiki
- Start battery by applying 12V to it, and CAN H-L pins to a CAN reader / LilyGo 
- Get a CAN log of the battery in standalone operation
- Start mapping values!
- Create a battery .cpp and .h header, see the other batteries for how the structure should look like
- Figure out each value in the CAN messages needed for safe operation

Probably forgot some steps, but that is the basic gist of it!

dalathegreat, 2023-09-26:
@cgalpin The Huawei SUN2000 inverters can take 800V batteries (or 2x 400V packs)

cgalpin, 2023-09-27:
Interesting.  I now see the wiki entry, but didnt see any code for it.

I am a noob to this, but everything I look at still says 600V DC input max. What am I missing? I am in the USA, so it has to be single phase 120V AC out as well (afaik).

For example, this show 600V - https://midsummer.ie/pdfs/huawei-sun2000-2-6ktl-l1-datasheet.pdf

kalectro, 2023-10-13:
@cgalpin afaik the 3-phase Huawei SUN2000 inverters need a higher battery voltage than the 1-phase inverters. Thinking about it... how do they both work with the same battery (LUNA2000) 🤔

Platin21, 2023-10-15:
Just a question on that note the Pylon Tech stuff is already supported?

dalathegreat, 2023-10-21:
> Just a question on that note the Pylon Tech stuff is already supported?

Yes, Pylontech CAN protocol is implemented!

stretchkerr, 2023-11-21:
Hi Dala, I don't see the BYD Atto 3 battery on the list of compatible batteries, yet the repo title is "BYD Battery Emulator..."

What am I missing?

I have a 60kWh Atto 3 battery pack I'd love to integrate.

stretchkerr, 2023-11-21:
> @stretchkerr The project started out as emulating only BYD HVM (solar batteries). Now the project can emulate lots of other brands.
> 
> The BYD Atto3 is not a solar battery, it is an automotive battery. To integrate it, follow the steps as explained here: [#51 (comment)](https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/issues/51#issuecomment-1685365046)
> 
> Lots of work!

Thanks for the explanation. BYD solar != BYD Auto.
Currently trying to track down a pinout for the Atto3 CAN connector. Will provide an update if I find anything.

cgalpin, 2023-12-29:
> @cgalpin afaik the 3-phase Huawei SUN2000 inverters need a higher battery voltage than the 1-phase inverters. Thinking about it... how do they both work with the same battery (LUNA2000) 🤔

Yes everything I am seeing shows that 3-phase inverters can take >600V input (including the battery), but 1-phase maxes out at <500V. 

I don't know if the battery is the same for both applications.

Can anyone confirm there is a single phase inverter that can be paired with a ~800v battery, and if so which model(s)?

KarlZeilhofer, 2024-06-19:
> Can anyone confirm there is a single phase inverter that can be paired with a ~800v battery, and if so which model(s)?

I don't think, single phase inverters can take such high voltages. The inverters internal rail voltage should be about 350VDC for 230VAC output. For a 3 Phase 230/400VAC the rail voltage would be 650VDC. 

Are there any news on connecting 3rd party battery systems to a Huawei Sun2000 inverter?

dalathegreat, 2024-07-01:
I am closing this issue. The answer to the original question posted in this thread, "How to add new battery" has a dedicated wiki thread here with info: https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-battery-integration

closed 2024-07-01 by dalathegreat