dalathegreat/Battery-Emulator #58: Incorrect data reported back to Sungrow over CAN

swaglordmcmuffin, 2023-09-17:
Hi, it appears that Sungrow does not detect SOC and charging power when running even though the system recognizes the emulated battery.

Has anyone experienced anything similar? I can see the SOC, SOH and some voltage but the inverter does not display the power going to battery - but instead it lists it as house consumption (9kW consumption on forced charge).

Is there something missing in the communication when the Lily's expected to send something back to the inverter, causing all power from grid to be interpreted as consumption?

dalathegreat, 2023-09-17:
@swaglordmcmuffin What settings? What battery? What does the USB serial log look like?

 We need more info, lots of people use the Sungrow inverters successfully, so most likely some small error somewhere :)

swaglordmcmuffin, 2023-09-17:
Hi! Well, I can see that I left out some important stuff here since I had to rewrite my post due to the browser crashing 😂

We're using Leaf 24kWh batteries, connected to SH10RT using CAN. I'll try to extract some logs next time I'm on site.

But we can confirm that this is features that's working for other people? If so, the problem is probably 100% at our end. I'll dig into it next opportunity I get!

dalathegreat, 2023-09-17:
Yes, I can confirm that we have multiple people on the Discord server that use Sungrow inverters.

Try to get more info!

dalathegreat, 2023-10-16:
@swaglordmcmuffin Did you ever get more info? I suggest we close this issue otherwise!

closed 2023-10-21 by dalathegreat