dalathegreat/Battery-Emulator #70: Maxus eDeliver3 battery pack

DaanNL76, 2023-10-23:
I have a Maxus eEdeliver 2022 battery pack, any information on that one available yet?

dalathegreat, 2023-10-23:
@DaanNL76 Unfortunately not, but feel free to start investigating it! See https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/issues/51 for info on how a new battery is added

DaanNL76, 2023-10-25:
Hi Daniel,
Thnaks for your very quick response. Do you have a bit more detailed description on how to get my work started?

M4GNV5, 2025-03-22:
I am currently in the progress of buying an eDeliver 3 for a camper conversion. I did look at some of the technical details to evaluate how hard it would be to use a SMA Sunny Boy Storage for 230V on the go (rather than an Anker powerstation, what most campers conversions use).

Regarding CAN bus contents: If everything goes well I might be able to supply CAN dumps in April. However the [OVMS](https://www.openvehicles.com/) project already has support for the eD3, at least for some basic signals like SOC, SOH, voltage, amps:
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/tree/master/vehicle/OVMS.V3/components/vehicle_maxus_edeliver3
However at first glance they seem to use the diagnostic CAN bus, which is only connected to the gateway, even though the OBD Port also includes the charge-CAN (on non standard pins)

Regarding wiring: Googling for `maxus edeliver 3 service manual filetype:pdf` turns up a 322 page manual including parts replacement tutorials and wiring diagrams.

Apart from the HV plug, the BMS seems to have two plugs "A" and "B", A seems to be responsible for charging, B for everything else.
I do not have pictures of those connectors yet, but the service manual includes pin numbers & labels for different pins.
CAN wiring (PT CAN = powertrain CAN):
[image]

HV Interlock on B pins 2 and 3: 
[image]

All other BMS connections:
[image]