dalathegreat/Battery-Emulator #86: Leaf 62kWh, SOC and GIDS freezes

StarkJohan, 2023-11-11:
On my Leaf 62kWh battery, SOC and GIDS shows very strange values after a few days of running grid support services through Checkwatt. SOC/GIDS is suddenly **not affected by slight charge/discharge** (1A) while battery and cell voltage remains correct and increases with low rate charge/discharge as well as larger loads. SOC and GIDS only changes if charge/discharge rate is significant.

In my setup this leads to SOC/GIDS of around 65% while the true charge of the battery is at the full 96%. It also means the true SOC can be pushed outside the software set limits. Bad news on more than one level.

This issue likely occurs because of the way Checkwatt controls the battery SOC with Solax inverters. They always aim for 65% which means the battery is constantly changing between charge/discharge at a very low level to stay at 65% SOC (at the inverter). The leaf BMS does not like this and bugs out after a few days. A reboot of the BMS resets the SOC/GIDS values but might put the true SOC outside the configured limits which is not good.

We need to improve software checks to make this, and similar scenarios impossible. Some basic checks are in place. This will avoid bad battery failures but does not truly solve the problem as the battery will be put in fault mode until manually rebooted. 
When installed in a car the BMS power is cut for two minutes every 24 hours. 

A simple timer or otherwise controlled 12V input could reboot the BMS on a certain interval, e.g. break 12V power 1 minute every 24h. The software needs to handle this scenario correctly to make sure everything continues to operate after a reboot. I have not yet properly tested this approach but it should work without any large code additions.
I would like an integrated solution much like the contactor control. If using contactor control and dual can the lilygo is out of available pins. If dual can is not needed, I suggest "borrowing" on of those pins. 
I'm working on an all-in-one hardware solution to replace the lilygo and I intend to make room for another 12V "channel" in addition to the three needed for contactor control.

_The fault mode could be delayed with a software mod to make the 64-65-66% percent interval to be virtually larger but this is a very specific addition and most likely not useful in any other setups._

_Values from battery
Real SOC%: 53% , GIDS: 355 (x77Wh) , Battery gen: 2 , Max cell voltage: 4152mV , Min cell voltage: 4119mV , Cell deviation: 33mV Values going to inverter
SOH%: 94% , SOC% scaled: 42% , Voltage: 396V , Max discharge power: 30000W , Max charge power: 11000W , Max temp: 22°C , Min temp: 21°C 
BMS Status: Active, Idle, Power: -11880W_

dalathegreat, 2023-11-11:
@StarkJohan Thank you for the detailed bug report. From Nissans own "Battery_control_spec_ver1 0.pdf", they mention the following regarding the BMS power:
[image]

So it seems like Nissan themselves did not mean for the BMS to be in continous operation, and will require a reboot every day. This reboot with the 6min30sec OFF time should be implemented in your solution.

I have added a "SOC% plausible?" check in this branch, feel free to test it out. It will help catch any dangerous situation, and halt operation incase values are not calculated correctly. https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/compare/main...bugfix/LEAF-frozen It is not an end-all solution, but better than nothing!

Also very interesting that this only seems to glitch out when the direction of charge/discharge changes rapidly. My own 30kWh LEAF pack has been smooth sailing for over 2months uptime! But very good that we get this into the software

StarkJohan, 2023-11-11:
The shortest possible downtime is desirable when using Checkwatt and similar services. A very short power cut seems sufficient as I've done roughly 10s which has worked fine so far. 6,5 minutes seems very cautious but I suggest letting this be the default/recommended value. It should be easy to adjust though.

At this point I assume most of us use 12VDC continuously on both BAT and IGN but the "IGN to be off" part suggests maybe we should handle this differently?

Before I hooked my system up to Checkwatt it was running for a couple of months without ever running into this issue so it seems very likely to be related to the very numerous cycles.

dalathegreat, 2024-12-29:
@StarkJohan Better late than never, proper BMS restart is now available in PR #706 so we can soon close this bug!

StarkJohan, 2024-12-29:
Excellent! Will definitely try when I'm back at the battery location.

dalathegreat, 2025-01-15:
Closing as completed, the PERIODIC_BMS_RESET is implemented to combat this :heavy_check_mark:

closed 2025-01-15 by dalathegreat