dalathegreat/Battery-Emulator #116: Compilation error: 'CAN_TX_IDX' undeclared (first use in this function)

BEMSman, 2023-12-20:
Hello,
          Came across this project and looks great, giving it a try I am encountering an issue, firstly I am using Arduino IDE 2.2.1 and the latest version 4.1.0 of BYD-Battery-Emulator-For-Gen24, I have tried different battery / fake config by uncommenting choice in USER_SETTINGS.h and still receive the below error, have checked config.h in src/devboard and the pin assignments are present and correct, I believe there is probably something I need to do but cannot find any pointers any help would be welcome, thanks for any assistance:

C:\Users\Chris\Downloads\BYD-Battery-Emulator-For-Gen24-4.1.0\BYD-Battery-Emulator-For-Gen24-4.1.0\Software\src\lib\miwagner-ESP32-Arduino-CAN\CAN.c: In function 'CAN_init':
C:\Users\Chris\Downloads\BYD-Battery-Emulator-For-Gen24-4.1.0\BYD-Battery-Emulator-For-Gen24-4.1.0\Software\src\lib\miwagner-ESP32-Arduino-CAN\CAN.c:181:44: error: 'CAN_TX_IDX' undeclared (first use in this function)
  181 |         gpio_matrix_out(CAN_cfg.tx_pin_id, CAN_TX_IDX, 0, 0);
      |                                            ^~~~~~~~~~
C:\Users\Chris\Downloads\BYD-Battery-Emulator-For-Gen24-4.1.0\BYD-Battery-Emulator-For-Gen24-4.1.0\Software\src\lib\miwagner-ESP32-Arduino-CAN\CAN.c:181:44: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\Chris\Downloads\BYD-Battery-Emulator-For-Gen24-4.1.0\BYD-Battery-Emulator-For-Gen24-4.1.0\Software\src\lib\miwagner-ESP32-Arduino-CAN\CAN.c:186:43: error: 'CAN_RX_IDX' undeclared (first use in this function)
  186 |         gpio_matrix_in(CAN_cfg.rx_pin_id, CAN_RX_IDX, 0);
      |                                           ^~~~~~~~~~

exit status 1

Compilation error: 'CAN_TX_IDX' undeclared (first use in this function)

BEMSman, 2023-12-20:
UPDATE!!
I just removed ESP32 by Espressif Library ver 3.0.0-alpha2 and rolled back to ver 2.0.11 and can now compile without error.

dalathegreat, 2023-12-20:
Great, once the library leaves ALPHA, we can update the repo. They wanted to rename CAN to TWIA which messes up a few things. One of my Patreons posted on Github same thing when they tried the ALPHA;

_Error: cannot find file "esp_intr.h"
New file name is "esp_intr_alloc.h"
Error: no definition for CAN_TX_IDX, CAN_RX_IDX
New defines are TWAI_TX_IDX and TWAI_RX_IDX, not sure what file is needed now, maybe "soc/gpio_sig_map.h" but using hard coded values from that file (found on github) works well. TX is 123, RX is 94_

dalathegreat, 2024-04-23:
The merged PR https://github.com/dalathegreat/Battery-Emulator/pull/262 makes the users aware that only 2.0.11 works at the moment!

closed 2024-04-25 by dalathegreat