dalathegreat/Battery-Emulator #118: Tesla battery power wrong value on the webpage

MarkoBolt, 2024-01-02:
Battery power has the wrong value shown on the webpage. 

[image]

Could the following line cause an overflow?

https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/blob/430da0e0bd7f35e1cdde4469e814cecd5ef30641/Software/src/battery/TESLA-MODEL-3-BATTERY.cpp#L205C3-L205C68

dalathegreat, 2024-01-02:
Nice find @MarkoBolt , I pushed an attempted fix to this branch: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/compare/main...bugfix/tesla-power

Wanna test it? :)

MarkoBolt, 2024-01-02:
Tested it, dot not work ok yet.

Tested with the following line ant then it seems to be ok. 

`power = ((volts/10) * amps); `

Value is too large for the variable power, causing an overflow in the variable power. Next to that it should be 10 times less, so the divide by 10 helps in this case, so it is scaled ok.

dalathegreat, 2024-01-02:
@MarkoBolt Thanks! Fixed in https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/pull/119/commits/6b06132607e9e00539ac3b27a95d9786567132ab

closed 2024-01-02 by MarkoBolt