dalathegreat/Battery-Emulator #128: JK BMS integration

kochikame, 2024-01-08:
Hi,
What would be need to be integrated with a JK BMS DIY battery?
What is the part I have to change in code?
I would like to connect the Fronius inverter with my ome brew batteries
and I would like to share it too.
Thnaks in advance!

dalathegreat, 2024-01-09:
@kochikame 
JK BMS is for max 24S. This project is focused towards safe re-use of complete 96S 400V EV batteries as-is.

To use JK BMS, you would need to build a 96S pack with multiple JK BMSes at the same time. This is quite massive task to build and integrate. So I do not think it really suits the 400V use case here for a Fronius inverter. But this is open source, so feel free to attempt it :)

kochikame, 2024-01-09:
Thnaks for the feedback othewise it could be a good challenge
Onother question... The project is for GEN 24 Fronius inverters but could it work on other inverters with BYD ready compatibility ?
Thanks and congrats for the project

alexsve1990, 2024-01-14:
I'm working on a JK BMS integration for this application. I'm almost finished.

kochikame, 2024-01-15:
@alexsve1990 I suppose you are programig  a class to interface with JK BMS and collect all the info of several JK BMS and pack them as if there were from LEAF like?
If I can help let me know. I am very interested in this optoin due I can not find a decent second hand  LEAF 24kw

kochikame, 2024-02-09:
I also almost finish a JK BMS develoment, consists in several JK BMS 24 packs to get almost 300 volts
I amb implementing JK protocol individualy for each pack with a M5STACK CORE
And then I will try to gather all the info in the LilyCAN and reuse the code to transmit all escential data to the inverter
I will keep on posting...

Elektrodude, 2025-05-22:
Any update here?