dalathegreat/Battery-Emulator #134: Weird characters in Serial Monitor

Jhonyboy72, 2024-01-13:
First of All: Thanks for the great, great work.

today i uploaded the Software to LilyGo-Board. Everthing just fine. 
But wenn ik open the Serial Monitor i only see weird characters.
Anyone else had this problem before and knows how to solve it?

Thanks in advance

[image]

LevinSwe, 2024-01-13:
Hi,
Change your baudrate to 115200 👍
Regards
[image]

Jhonyboy72, 2024-01-13:
Works just fine now! Thank you very much!

closed 2024-01-14 by dalathegreat