dalathegreat/Battery-Emulator #138: conflicting declaration

smotek, 2024-01-18:
Hello,
first of all, I have to thank you for the perfect work.
So far I've only tried compiling in the IDE.
If I don't activate the webserver, it works fine.
If I activate the Webserver, 
#define WEBSERVER 
I get a compilation error:
In file included from BYD-Battery-Emulator-For-Gen24-main\Software\Software.ino:18:
BYD-Battery-Emulator-For-Gen24-main\Software\src\devboard\webserver\webserver.h:21:17: error: conflicting declaration 'uint16_t ms_status'
 extern uint16_t bms_status;                  //Enum, 0-5
                 ^~~~~~~~~~
In file included fromBYD-Battery-Emulator-For-Gen24-main\Software\src\inverter\INVERTERS.h:33,
                 from BYD-Battery-Emulator-For-Gen24-main\Software\Software.ino:9:
BYD-Battery-Emulator-For-Gen24-main\Software\src\inverter\SERIAL-LINK-TRANSMITTER-INVERTER.h:20:16: note: previous declaration as 'uint8_t bms_status'
 extern uint8_t bms_status;                   //Enum, 0-5
                ^~~~~~~~~~
In file included from BYD-Battery-Emulator-For-Gen24-main\Software\Software.ino:18:
BYD-Battery-Emulator-For-Gen24-main\Software\src\devboard\webserver\webserver.h:22:17: error: conflicting declaration 'uint16_t bms_char_dis_status'
 extern uint16_t bms_char_dis_status;         //Enum, 0-2
                 ^~~~~~~~~~~~~~~~~~~
In file included from BYD-Battery-Emulator-For-Gen24-main\Software\src\inverter\INVERTERS.h:33,
                 from BYD-Battery-Emulator-For-Gen24-main\Software\Software.ino:9:
BYD-Battery-Emulator-For-Gen24-main\Software\src\inverter\SERIAL-LINK-TRANSMITTER-INVERTER.h:21:16: note: previous declaration as 'uint8_t bms_char_dis_status'
 extern uint8_t bms_char_dis_status;          //Enum, 0-2
                ^~~~~~~~~~~~~~~~~~~
                
 I don't dare to fix it.
How is that correct?

I still have one question:
If I define
#define SERIAL_LINK_TRANSMITTER
#define TESLA_MODEL_3_BATTERY
after connection, the BMS is activated.
If I connect the AC connector to the battery, will the AC charging work?

Thank you

dalathegreat, 2024-01-18:
Please do not use main directly, it can be in a state of development.

Use a RELEASED version of the software, via the releases tab:
[image]

Latest stable version is 4.3.0: https://github.com/dalathegreat/Battery-Emulator/releases/tag/v4.3.0
Click the  "Source code (zip)" at the bottom.

smotek, 2024-01-18:
It's ok now. It was the work-in-progress version.

Please, do you know how it is with the AC charging?
I am interested in this because it is possible to charge the battery via AC before storage.

dalathegreat, 2024-01-18:
Alll solar inverters can charge the battery via DC from the grid. So no need to manually charge it to a certain SOC% before using it in storage mode.

smotek, 2024-01-18:
I understand that.
There will be 2x lilygo and Solax G4.
It was just a question of whether it was technically possible.

dalathegreat, 2024-01-19:
Closing this issue (sincec it is not an issue :) )
If support is needed, I recommend the https://secondlifestorage.com/ forum, or joining my Patreon Discord server!

closed 2024-01-19 by dalathegreat