dalathegreat/Battery-Emulator #139: Battery has stopped reporting to Lilygo

mondc95ba, 2024-01-21:
Hi Dala,

Apologies if this is the wrong forum to raise this issue but I have a AZE0 24kwh leaf battery and lilygo working and reporting to the fronius inverter.

I've been waiting for my installer to complete the upgrade to the gen24 plus from his end, so the battery hasn't been charging or discharging. Currently using the manual pre-charge, contactor method until the SSR arrive.

As of recent the lilygo is throwing error code 7, no matter the restart sequence. Do you know if this is an issue with the battery going into a sleep mode due to not seeing any discharge/charge for a week whilst being on and communicating?

I have also measured resistance when the 12v is off and the lilygo unpowered and it reads 60 ohm.

mondc95ba, 2024-01-21:
[image]
[image]

Cabooman, 2024-02-02:
I would suggest you power just the BMS and hook up your choice of CAN dongle as a first step. Current draw at ~12.7 V should be around 0.30-ish Ampere if I remember correctly. If you don't see any current draw or chatter on the CAN bus, you might have an LBC issue as you say

dalathegreat, 2024-02-06:
I see you have a post on the diyelectriccar.com forum going, and extensive battery surgery ongoing aswell. I think it is best to continue discussion there, instead of the issue tracker for Battery-Emulator. Closing this issue!

closed 2024-02-06 by dalathegreat