dalathegreat/Battery-Emulator #150: Kostal Plenticore RS485

HWBHWB1, 2024-02-01:
I have the problem that my Kostal Plenticore Plus can only process the BYD battery via the RS485 interface. Unfortunately, the byd modbus that is already included in the software does not work. what can I do?

dalathegreat, 2024-02-03:
Should work with Modbus RTU, make sure you have high voltage on the BAT input when setting up:
[image]

HWBHWB1, 2024-02-03:
I tried with Modbus but I always get the error "ModbusServerRTU.cpp [252] serve: RTU received: E5 - packet length error" and the voltage of the Renault Kangoo battery always shows 6300 volts

dalathegreat, 2024-02-03:
@HWBHWB1 The problem is not that the Modbus is not working, the problem is that the Renault Kangoo battery has not been tested enough. Do you have the 22kWh or the 33kWh battery?

HWBHWB1, 2024-02-03:
I have the 22kwh battery from 2013, the connection seems to be working, I have replaced the cable several times and have now replaced it with cat6

HWBHWB1, 2024-02-03:
I get the modbus error even when I select the fake battery

dalathegreat, 2024-02-03:
I also get occasional "ModbusServerRTU.cpp [252] serve: RTU received: E5 - packet length error" when using the Modbus, without any issues. 

I think we need to investigate the CAN side of the Kangoo. Please join the Discord server, let's continue chatting there! https://discord.gg/YDKHsWxF

rand12345, 2024-02-05:
> I have the problem that my Kostal Plenticore Plus can only process the BYD battery via the RS485 interface. Unfortunately, the byd modbus that is already included in the software does not work. what can I do?

The inverter supports BYD and PylonTech Force H2 via CAN bus. This repo has fully tested ZE40 (Kangoo) protocol and is 2 years mature with reliable and inexpensive hardware.

https://github.com/rand12345/toucan_stm32f407/releases/tag/release

HWBHWB1, 2024-02-05:
Well, I would like to do it with my lilygo setup too, it has to be manageable

HWBHWB1, 2024-02-05:
The Kostal seems to automatically recognize the CAN protocol but it just has to be the right one

rand12345, 2024-02-06:
I initially developed a system using the LilyGo and abandoned it. The build quality is poor, the PCB corrodes, and the ESP32 is not a good CAN performer.

dalathegreat, 2024-02-06:
Made some great progress with @HWBHWB1 , the battery displays all data OK, but the inverter integration is for some reason misbehaving. Very strange, since others have reported the Kostal inverters to work great. What makes it even stranger, is that the inverter shows 255% SOC when using PylonCAN as an alternative. I am sure we will get it solved soon.

@rand12345 I do not think promoting your solution is bringing much to the table here. We are working on a custom Battery-Emulator PCB with multiple CAN buses, CAN-FD support, and built in relays for contactor closing procedure. If you are not helping the discussion by posting suggestions on what to check next, I suggest not posting at all.

dalathegreat, 2024-11-27:
Closing as completed. Kostal RS485 support added in release v7.8.0

closed 2024-11-27 by dalathegreat