dalathegreat/Battery-Emulator #156: Supporting Growatt MOD inverters

SelverC, 2024-02-05:
Hi Dala!

Great effort on this, is there anyway to help out and try to add support for Growatt Inverters?

SelverC, 2024-02-06:
Hi @dalathegreat,

I already own a set, so do you have sniffer to recommended?

dalathegreat, 2024-02-06:
@SelverC , Ebay/amazon/aliexpress has inexpensive (<5€) dongles you can get. Just search for "Modbus USB"

dalathegreat, 2024-02-07:
Also @SelverC , please post exact model numbers of the Inverter / Battery ! :)

Palmtree1455, 2024-02-10:
I'm wondering about the Growatt MIN series for the US.  I'll be interested in your findings.

edmundasa, 2024-03-18:
@SelverC any progress on those logs? I'm considering this inverter as I can get it cheaper than others. Fingers crossed that you can get them. Thank you both.

Palmtree1455, 2024-03-18:
I think I'm going to try the solis s6 instead.

tmlarsson, 2024-03-26:
Following this as I recently got my hands on one of these at a killer price.

Palmtree1455, 2024-03-26:
Where did you find the killer price?

tmlarsson, 2024-03-27:
@Palmtree1455 Bought a 8000w xh version second hand for 400 EUR locally in Sweden

Odin263, 2024-04-04:
Hello everyone, I would like to pursue this topic further as I also have a working system with Growatt components. The inverter is a "SPH 10000TL3 BH-UP" and the battery is "ARK 7.6H-A1", which consists of 3 battery modules "ARK 2.5H-A1" and the BMS "HVC 60050-A1". I would still have to check how I can get logs from there, but it will certainly be possible. Unlike the use case described, I don't want to have a battery stationary at home, but I would like to use the battery from a NISSAN LEAF via the CHAdeMO connection as a buffer for the existing 7.5kWh battery storage, but also have the option of simply closing the car quickly charging (10kW with this inverter), whereby I can switch the CHAdeMO connection on or off via the Growatt app (in this case it replaces the Growatt battery storage), which also works depending on conditions such as the power of the PV modules. If I interpret it correctly, can it be implemented in the same way as the stationary battery storage system described?

tmlarsson, 2024-05-30:
Think I maybe have found a protocol specification for these inverters.

[Growatt PV Inverter Modbus RS485 RTU Protocol v120.pdf](https://github.com/ai-republic/bms-to-inverter/files/12640748/Growatt.PV.Inverter.Modbus.RS485.RTU.Protocol.v120.pdf)

dalathegreat, 2024-06-02:
@tmlarsson Looks like 48V Modbus registers, would be better to get a High voltage communication log from a running system. Without that, it's almost like guessing everything!

dalathegreat, 2024-06-09:
To continue with the Growatt integration, we need logs from someone that has a functional Growatt + Growatt ARK battery setup. If we get those logs, we can start to develop support for Growatt emulation. Please let us know if you know someone with a "ARK XH Battery System(7.68kWh 25.6kWh)" battery willing to log!

ErikssonPer, 2024-12-02:
Hi, I've got growatt SPH10000 and a 10kWh ark battery that I can pull some logs from with either a kvaser canusb or waveshare canhat.

Should i make log of a full charge/recharge cycle or speciific moments like near the end of charge when the current is beginning to ramp down or a power up sequence?

Happy to help out.

dalathegreat, 2024-12-02:
Hi @ErikssonPer , that would be great! I have instructions here on what is needed: https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-inverter-integration

ErikssonPer, 2024-12-03:
Did some logs today.
Included a screenshot of how the bms is ramping down the current step by step near the end of charging.
[Start of inverter and then battery.txt](https://github.com/user-attachments/files/18000174/Start.of.inverter.and.then.battery.txt)
[Charging to full.txt](https://github.com/user-attachments/files/18000183/Charging.to.full.txt)
[Charging.txt](https://github.com/user-attachments/files/18000193/Charging.txt)
[discharging.txt](https://github.com/user-attachments/files/18000196/discharging.txt)
[Inverter shutdown.txt](https://github.com/user-attachments/files/18000197/Inverter.shutdown.txt)
[Battery Shutdown.txt](https://github.com/user-attachments/files/18000198/Battery.Shutdown.txt)
[image]

ErikssonPer, 2024-12-24:
Who's working on Growatt integration?
I found growatts specs for the high voltage, low voltage CAN protocols and RS485 protocol.
I have checked HV CAN against the logs I've taken from my SPH inverter and it matches.

dalathegreat, 2025-01-02:
@ErikssonPer Could you post the files here?

ErikssonPer, 2025-01-02:
[Communication.protocol.of.Growatt.high.voltage.battery.CAN.V1.10.3.pdf](https://github.com/user-attachments/files/18294197/Communication.protocol.of.Growatt.high.voltage.battery.CAN.V1.10.3.pdf)
[Growatt-BMS-CAN-Bus-protocol-low-voltage-V1.04-1 (1).pdf](https://github.com/user-attachments/files/18294198/Growatt-BMS-CAN-Bus-protocol-low-voltage-V1.04-1.1.pdf)

dalathegreat, 2025-01-10:
Thanks @ErikssonPer , I will get started on this!

dalathegreat, 2025-01-12:
@Star442 

HV protocol added here: https://github.com/dalathegreat/Battery-Emulator/pull/788

dalathegreat, 2025-01-13:
Do note that the communication protocol posted by @ErikssonPer is only applicable to inverters compatible with the HVC 60050-A1 BMS. And accorning to Growatt documentation that is SPH/SPA 4000-10000TL3 BH-UP inverters. I have updated [the Wiki](https://github.com/dalathegreat/Battery-Emulator/wiki/Inverter:-Growatt-HV), but someone feel free to add more info there!

ErikssonPer, 2025-02-06:
@dalathegreat 
Did a quick and dirty test with it set as a fake battery matching the actual voltage my batterypack has right now and it is able to charge and discharge the battery and responds to the max charge and discharge currents set in the software also.

Looks like a win!

Next will be to run it all the way to max and min SOC and check that it reacts correctly to the limits set.

[image]

dalathegreat, 2025-02-06:
Be careful running fake battery, best to switch to actual battery protocol before further testing! Nice to see it works!

dalathegreat, 2025-02-16:
Since both [Growatt HV](https://github.com/dalathegreat/Battery-Emulator/wiki/Inverter:-Growatt-HV) and [Growatt LV](https://github.com/dalathegreat/Battery-Emulator/wiki/Inverter:-Growatt-LV) have been added now, let's close this as completed, and continue in new issue threads incase there are any problems. Thanks again to all that contributed to getting these protocols in 🙌

closed 2025-02-16 by dalathegreat