dalathegreat/Battery-Emulator #161: Pylontech Force H2 as Battery

nils-ossenbrink, 2024-02-07:
Hi,

I would like to use the battery mentioned above with my Fronius GEN24 (I bought it together with a GoodWe inverter before I found this nice project :-( )
Any hints where I can find some more detailed documentation about the can-protocol (in addition to PYLON-CAN.h/cpp)? 

Thanks!

Cheers
Nils

dalathegreat, 2024-02-07:
Hi @nils-ossenbrink 
DBC file available here for Pylon protocol: https://www.eevblog.com/forum/programming/pylontech-sc0500-protocol-hacking/

flo-tschi, 2024-03-31:
Hello Nils!

Did you had any success? I would also need this implementation because I want to connect my diybmsv4 to my fronius inverter. 

best
Florian

nils-ossenbrink, 2024-04-01:
Hi, unfortunatrly not. I started to implement but my time ressources are
limited :-( The pylontech is connected to a goodwe Inverter now and is
working quite well. My aim is to run it with the fronius but I currently
don't know when I will have time for this project...

flo-tschi, 2024-04-02:
That’s too bad!

I will try what I can do - but actually it’s quite a long time since I last programmed anything.

dalathegreat, 2024-04-25:
@nils-ossenbrink 
@flo-tschi 
This has been added to 5.8.dev now in PR https://github.com/dalathegreat/Battery-Emulator/pull/265

Closing this ticket as completed!

closed 2024-04-25 by dalathegreat