dalathegreat/Battery-Emulator #179: QUESTIONS

kochikame, 2024-02-13:
Hi
regarding these defines
does anyone know how exactly the function of these features?
thx in advance

//#define CONTACTOR_CONTROL     //Enable this line to have pins 25,32,33 handle automatic precharge/contactor+/contactor- closing sequence
//#define PWM_CONTACTOR_CONTROL /
//#define SERIAL_LINK_RECEIVER  //Enable this line to receive battery data over RS485 pins from another Lilygo (This LilyGo interfaces with inverter)
//#define SERIAL_LINK_TRANSMITTER

dalathegreat, 2024-02-13:
@kochikame 
Contactor control / PWM control is useful for LEAF batteries, see Automatic Control section for explanation of this feature: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/wiki/Nissan-LEAF-battery#automatic-control- 

The Serial Link is for Dual Lilygo funcitonality: https://github.com/dalathegreat/Battery-Emulator/wiki/Double-LilyGo

closed 2024-02-14 by dalathegreat

mondc95ba, 2024-02-18:
no circuit diagram for the SSR circuit, propose using the below if suitable?
[image]

dalathegreat, 2024-02-19:
Massive thanks @mondc95ba , added to the Wiki! https://github.com/dalathegreat/Battery-Emulator/wiki/Nissan-LEAF-battery#automatic-control-