dalathegreat/Battery-Emulator #180: MQTT bug in voltage?

MarkoBolt, 2024-02-13:
I am testing 5.2.0. MQTT battery voltage is 0V in my Home Assistant. 

Seems to be a bug in the following line. 

https://github.com/dalathegreat/Battery-Emulator/blob/a238c8b92519d910271bb8c6ffa331fed8ce235e/Software/src/devboard/mqtt/mqtt.cpp#L186

"battery_voltage"  is an integer. Divide it by 10.0 results in no voltage data send in my setup.

The following works for me since I am not interested in a voltage decimal.
`             ((float)cell_min_voltage) / 1000, battery_voltage / 10);`

Cabooman, 2024-02-14:
Seems like it's something unrelated to mqtt, rather some other part of the code interfering with this variable. Perhaps another declaration? From 10 minutes of testing it feels quite random, which could be explained by some compile time timing dependency... the best type of bug :)

Some testing below
-----

Initially tested `((float) battery_voltage) / 10.0` as in all other places. Still 0.

Tested with:
```
...
"  \"battery_voltage\": %.3f,\n"
"  \"battery_voltage2\": %d,\n"
"  \"battery_voltage3\": %.3f\n"
... using
((float)battery_voltage) / 10.0,
battery_voltage,
static_cast<float>(battery_voltage) / 10.0
```
which resulted in:
```
350.000
3500
350.000
```

dalathegreat, 2024-05-12:
Closing this as unable to reproduce?