dalathegreat/Battery-Emulator #187: Temperature derating

Cabooman, 2024-02-16:
It would be nice to limit charge/discharge power based on the battery temperature, e.g. ramping the power from the stated BMS power capacity at 40 degrees C  to 0 W at 50 degrees C. Similarly at low temperature, ramp from BMS max power at -10 and 0 W at -20

Tosood, 2024-03-06:
I have a similar wish regarding the temperature rating or reduced power related to temperatures below 20 degC. Using Goodwe 30 kW inverter with simulated 2*BYD HVM batteries. 
Ideal would be if the temperature could be "translated" to 21 degC when in the range of -10 to 19,9 degC ? This would allow us to run full power when needed. Without having to heat the batteries at all times.

havrla, 2024-10-27:
How do you use batteries?
Battery management I3 continuously regulates the recommended charge and discharge value.
Keep in mind that at -10 celsius you cannot fully power  charging, the battery will be destroyed.
As for regulation according to temperature - this is determined by battery management. So your request is duplicated and actually unnecessary.
Only the battery management inside the battery has the correct values ​​for setting the limit.

dalathegreat, 2024-12-29:
Yeah I agree with havrla on this, the derating is automatic based on the BMS inside the battery. Derating due to temperature is only missing for those few integrations that are using Estimated charge/discharge limits (Tesla/RJXZS BMS spring to mind).

So I think instead of making our own derating, it would be better to implement the derating inside those few integrations that aren't derating.

closed 2025-01-02 by dalathegreat

Tosood, 2025-02-14:
Hej Daniel,
Vi har fått ett erbjudande om att köpa ett 100 kWh Porsche batteri av någon
som arbetar på en verkstad. Jag ser att det inte finns någon som har testat
det ännu. Tror du att det är på gång eller är du intresserad av att köpa?
Det finns motorer och växellåda med.
Hälsningar Tor Söderlund
+46761360693

dalathegreat, 2025-02-14:
Hej @Tosood 
Please create a new issue for this (you are commenting under a temperature derating closed issue)

AFAIK nobody has any Porsche logs, so getting a 100kWh battery to work will require significant amount of effort. Expect a few months of development to get one working! See https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-battery-integration

But yeah, new issue is best to go forward!

Tosood, 2025-02-14:
No Worries I just wondered if you wanted to buy it for cheap but seems like
it’s not intresting!
Have a nice weekend!
Tor