dalathegreat/Battery-Emulator #192: Cell monitor html formatting

BEMSman, 2024-02-18:
It would be great if it is possible to add function to be able to adjust the number of coulombs across the page to make it easier to know at a glance where an offending cell is:

For example Renault and Nissan have 8 cells per module, BMW, KIA and Hyundai have 12 cells per module.
so my thoughts are can you make a coulomb count of between 4 - 12

dalathegreat, 2024-02-25:
Even better, Cabooman is working on graphs:
[image]

BEMSman, 2024-02-25:
That sounds even better, that graph I assume is one line each cell, thats good as you will see where the offending cell is at a glance.
I have played around with the settings and have managed to get 12 cells wide, however its hard coding.
[image]

matvey83, 2024-02-26:
Graph is a great idea!   That's awesome.  But also if we know the cell arrangement in packs, we can also provide some kind of a visual that would help see how cell groups are arranged in modules.  For example, NCM model 3 batteries, it would be useful for user to know that cells are grouped into modules as 1-23, 24-48, 49-73, 74-96.   And on 108 cell LFP packs, they are grouped into 1-26, 27-54, 55-82, 83-108.   It would be helpful to see at a glance if the issue can be resolved by replacing one module or not.   For packs with more modules this would be even more helpful.

dalathegreat, 2024-07-01:
Closing this as completed, cell graphs was implemented way back earlier this year!

closed 2024-07-01 by dalathegreat