dalathegreat/Battery-Emulator #202: Hooking This Up While Still In The Car

tdashmike, 2024-02-22:
Has anyone attempted to have connectors put in the car so I can still drive the car and hook up the power when I need it?

lewurm, 2024-03-08:
I guess the best case scenario would be that it just works over a CCS connector. Here are some experiments documented that try to discharge via CCS: https://openinverter.org/forum/viewtopic.php?t=3551

With the battery emulator on the other side, an inverter could be used to drive a charging session. While I firmly believe _charging_ should be possible with a setup like this (would already be cool tbh), I'm afraid that most cars would abort the session after a few minutes of discharging is detected.

I hope I'm wrong though 😉 I got a CCS connector from Aliexpress, now I need to find some time to reproduce a setup like described in the thread above.

We need more experiments on that front 🙂

tdashmike, 2024-03-09:
I'm looking at getting a leaf because it has a CHAdeMO connector. If I can connect the emulator to the CANBUS through the connector I should be able to enable the contactors for drawing power out. I do not need to charge it through the CHAdeMO port. It would be nice to be able to use the leaf in an emergency. I can have a solar inverter hooked up to the house to convert the power.

https://www.evcreate.com/chademo-fast-charging-in-diy/

timol122, 2024-04-16:
Fully agree that this would be interesting. Ultimatly it could become a V2G solution with a compatible hybrid solar inverter. 

What are the open issues, risks and blockages towards this goal, anyone has the overview?

bnolan, 2024-07-17:
My 40kwh nissan leaf is going to be parked up in the sun for months at a time and I am charging it with 500w of solar on the roof. I'm currently using an ecoflow delta battery + 8 amp charger, but I would like to use a fronius to give a direct PV -> HV conversion. I would also like to be able to use the HV pack to provide on-site power.

I hope to do one of the following:

* Tee off the 6kw aux port into the fronius
* Tee off the main drive cable at the PDM into the fronius
* Use V2H with a Chademo plug

Have any of these been done before?

tdashmike, 2024-07-18:
> My 40kwh nissan leaf is going to be parked up in the sun for months at a time and I am charging it with 500w of solar on the roof. I'm currently using an ecoflow delta battery + 8 amp charger, but I [...]

I would like to do this but I don't want to spend the money right now. I'm also having trouble finding a compatible inverter in the US. All you really need to do is but a chademo plug and wire it up though. You can also print your own with a 3D printer.

https://www.aliexpress.us/item/3256807190705806.html

leomeno, 2024-11-04:
I've made some tests of V2X some while ago but not with the Battery-Emulator but using a Siemens PLC with their ECC modules, tested with more EVs, e.g. Hyundai Kona, Renault ZOE and Tesma M3. Even when their CCS doesn't officially suppport V2X, it works, with Renault and Hyundai you can simply start the charging session, the vehicle energizes the battery contactors and you can pull the power out, but with the Tesla you need to deliver some charging power first, othervise the charging process goes to error and stops. But after initial power delivery you can pull power from the vehicle. There also is this thingy with battery heating by Tesla with the DC charging, the vehicle heats the battery while DC charging, as far as I remember it pulls about 3kW (or maybe even 6kW, I don't remeber exactly), so when you start the charging session, the SOC level slowly drops even without any power consumption. But this behaviour can be disabled in the Tesla's service mode..

nicois, 2025-01-13:
I'm very interested in whether it will be possible to emulate a solar battery with a normal EV via its standard ports. 
I saw this video recently indicating SAIC/MG support for V2G: https://www.youtube.com/watch?v=Hi1_6_ShoxQ

With Fronius gen24 inverters doing all the heavy lifting of converting ~400VDC to AC and regulating the draw to match household consumption, it's "just" a matter of convincing the inverter that the car is a big battery.

lewurm, 2025-01-14:
> Can you expand on how you implemented that initial power delivery?

To answer this myself: It doesn't really need power, but the car wants to see some _voltage_.  I had success with a step-up converter; coincidentally the folks around getting the MEB battery up needed something similar, and came up with a nice mod to be able to control the voltage with a PWM instead of having to use the poti manually: https://github.com/dalathegreat/Battery-Emulator/wiki/High-Voltage-source

> There also is this thingy with battery heating by Tesla with the DC charging, the vehicle heats the battery while DC charging, as far as I remember it pulls about 3kW (or maybe even 6kW, I don't [...]

I know this was a problem in the past, but I haven't seen this behaviour on my car with my DC charger (up to 5kW, so maybe Tesla improved the logic there).

 > I saw this video recently indicating SAIC/MG support for V2G: https://www.youtube.com/watch?v=Hi1_6_ShoxQ
 
 interesting video!

>  it's "just" a matter of convincing the inverter that the car is a big battery.

Yes, I had some success on that "just" part.  I have a Kostal Plenticore inverter, that I was able to hookup with the battery of the car via its CCS port. More details can be found here: https://openinverter.org/forum/viewtopic.php?p=76707#p76707

My last big blocker are reliability issues around the communication with the car (the connection randomly times out, but I have some ideas to improve that). Once I figured that out I should have a rather useful setup, and then I'll do a full write-up.

There is one caveat, that the my car aborts the charging session when -15Ah have been discharged.  But you can just restart the charging session then, so not a big issue 😃 However this behaviour is not specified, but depends on the implementation/vendor. Not much data on this front, here is a small list: https://openinverter.org/wiki/Bidirectional_Charging#Compatible_cars

**TL;DR:** It's doable 🙂

lewurm, 2025-02-04:
> I know this was a problem in the past, but I haven't seen this behaviour on my car with my DC charger (up to 5kW, so maybe Tesla improved the logic there).

@leomeno some more insight on this: On the charger side I had `EVSEMaximumPowerLimit` set to 10kW, and haven't observed any battery heating even in cold weather (i.e. pack being at 10°C).  If I pump `EVSEMaximumPowerLimit` to 250kW the car starts to heat the battery with 6kW:

[image]

<sup><sub>Tested with car software version 2024.45.32.2</sub></sup>