dalathegreat/Battery-Emulator #203: Canbus failure on Sungrow SH10RS

Chair-Stealer, 2024-02-25:
I had it communicating for abit, and after a power cycle the liligo wasnt able to connect to the inverter again
I have two liligos setup and tried the second one with the same issue, the rs485 connection between the liligos is working correctly
Wiring appears to be correct according to the Sungrow manual
My setup is a tesla model 3 battery to a sungrow SH10RS inverter
Webserver fault reads CAN_TX_FAILURE

dalathegreat, 2024-02-25:
@Chair-Stealer For CAN TX failure troubleshooting, see https://github.com/dalathegreat/Battery-Emulator/wiki#can-wiring-troubleshooting

Chair-Stealer, 2024-02-25:
Cheers, the only on i havnt tried yet was the resistor test and shielding, ill see how i go for time and try it this this afternoon
Thankyou

Chair-Stealer, 2024-03-02:
Sorry for a late reply, work has been swamped and only got to working on it today. I am still having the same issues with no connection to the inverter, have gone through all troubleshooting steps outlined. the resistance between can H and can L is 65ohm. I have also tried the pushpin terminal and the RJ45 port on the inverter with no success. Have you got any more ideas that it could be. Cheers

dalathegreat, 2024-03-03:
@Chair-Stealer Sorry no clue! Lots of people with working Sungrow setups on the Discord server though, so it is most likely something minor hindering this!

Chair-Stealer, 2024-03-12:
@dalathegreat I have been in contact with Sungrow Australia with the issues I was having, After a phone call and email with support, they have said that Sungrow inverter no longer support BYD or Lgchem batteries. That would mean the new SH8.0RS and the SH10RS do not support anything but Sungrow HV batteries
[image]
So im not sure where to go from here
Thanks for you help

dalathegreat, 2024-03-13:
Thanks @Chair-Stealer , updated the wiki with your findings: https://github.com/dalathegreat/Battery-Emulator/wiki/Sungrow-inverters#compatible-sungrow-inverters

Seems we need to crack the Sungrow CAN protocol next to get the AU inverters working!

dalathegreat, 2024-03-16:
@Chair-Stealer I will close this issue, but please create a new issue like "Feature request, Sungrow CAN protocol" to keep tracking this!

closed 2024-03-16 by dalathegreat