dalathegreat/Battery-Emulator #205: Feature request: 800 Volt architectures.

tool-tony, 2024-02-25:
Used Hyundai Ionic 5 batteries can be very cheap compared to used Tesla batteries so is there an easy way to incorporate them into a house like these other ones? Thank you.

dalathegreat, 2024-02-25:
@tool-tony This is being worked on. Contactor closing on Kia EV6 800V batteries were completed today!

tool-tony, 2024-02-27:
> @tool-tony This is being worked on. Contactor closing on Kia EV6 800V batteries were completed today!

Awesome! In a few days, I intend to add the Max voltages from mfg spec sheets to the wiki since most don't have high enough ratings to handle these packs. How would you like that formatted?

tool-tony, 2024-03-05:
Is there an easy way to convert 3p to split 240? 

Other wise it looks like this is a single phase 208/240v output grid tie only with a Max input of 850vdc: http://www.ingeteam.com/Portals/0/Catalogo/Producto/Documento/PRD_1608_Archivo_ingecon-sun-1play-tl-u-m.pdf

I'd found this when browsing r/SolarDENY https://www.reddit.com/r/SolarDIY/s/qU5EpTeIQ8

Which links to your video funnily enough. 

Found in the linked thread: https://forums.mikeholt.com/threads/1000-volt-single-phase-inverters.133887/

Perhaps they didn't see that one post and focused only on the fronius? Which also seems to be 1000v capable but the normal Primo is not a hybrid so does not work while grid is down. 

I'd actually found a 75kwh Tesla Model 3 battery for much less than a Hyundai battery so I'd bought that instead. Just need to pick it up. Still interested in seeing this addition continue obviously! It's better for people to have more options.

dalathegreat, 2024-04-25:
Closing this issue as completed.
CAN-FD support for 800V E-GMP platform was added in 5.7.0
https://github.com/dalathegreat/Battery-Emulator/wiki/Hyundai-E%E2%80%90GMP-platform-(58.2-%E2%80%90-77.4-kWh)

closed 2024-04-25 by dalathegreat