dalathegreat/Battery-Emulator #207: MQTT sensor names in 5.3.0 changed

MarkoBolt, 2024-02-26:
The non cell MQTT sensor values have been changed in 5.3.0 resulting in new sensors via MQTT. 
In the new sensor names the lily board ID is present:
```
ESP32_1234567
```

kharnt0x, 2024-03-01:
@MarkoBolt the issue will be for those with multiple lilygos there will be a namespace conflict I think... the intent was to ensure we keep the sensor names unique

Cabooman, 2024-03-07:
At least for me and Mbolt (and probably others) the issue is that we'll get duplicate entities for a single setup in e.g. HA when using several lily's for testing purposes. Let's figure out a compile-time switch for differentiating (or not) when we're in dev mode again :)

StarkJohan, 2024-03-07:
We should take the step to MQTT discovery as this will make these kinds of problems easier to handle. As I see it we need to keep multiple scenarios in mind.

Most users will have one setup and might do some testing with different hardware. This scenario is best handled with generic sensor names and MQTT discovery. If the hardware changes, the sensor names stay the same.

Some users will use multiple setups with the same MQTT broker. In this case the sensors need to be unique on a system basis. I.e. some kind of system identifier is needed and I don't think the default esp host name is a viable solution.

I suggest adding a "hostname" variable in the user settings and swapping to MQTT discovery. This should solve both scenarios, including development.

Cabooman, 2024-03-07:
@StarkJohan Discovery is currently used for all MQTT data, but with said host name and the library that you’re not too happy with ;)

I was writing basically your post a few hours ago but didn’t manage to finish, so I agree with the scenario or use case perspective. Most will have one lily and one setup, some will then have two or more lilys with one setup and yet fewer will have some form of many lilys and many setups. If this was at work, we would solve it in the cabling, using one or more pins for system identification, but that doesn’t feel very feasible for a number of reasons.

In any case, “nice” entity names should be default. All ideas welcome!