dalathegreat/Battery-Emulator #210: more than one battery

Coolj1, 2024-02-28:
hi, and thanks a lot for this fantastic work!
is it possible to connect more then one battery (bmw i3)?

kharnt0x, 2024-02-29:
at the moment, I don't think so. one lilygo per battery.

dalathegreat, 2024-02-29:
I have 2x BMW i3 batteries in my garage, this is something we will develop support for later!

Coolj1, 2024-02-29:
ok thx

Coolj1, 2024-04-22:
hi dalathegreat,
there are 4 bmw i3 94ah batteries in my garage and a solax x3 g4 inverter connected to 10kW solar-modules.
Is there a way to connect all together...?
Maybe for each battery and the solax a separat esp with can-bus and the data from the battery-esp's to the solax-esp with rs485?

dalathegreat, 2024-04-23:
@Coolj1 Please start testing this branch with 2x batteries. Once we get 2x working, it will be easy to add more later!

nagyrobi, 2024-08-15:
https://hackaday.com/2024/08/13/from-vehicle-to-grid-to-diy-home-powerwalls/

Coolj1, 2024-08-15:
👍
great work!

dalathegreat, 2024-08-15:
Closing this as completed! :partying_face:

closed 2024-08-15 by dalathegreat