dalathegreat/Battery-Emulator #212: Testing Kostal Plenticore

jspv33, 2024-03-01:
I have connected a Plenticore Gen1 to a Leaf 30 KWH battery.
On the Plenticore: CAN H , CAN L, GND, 12 volts, 120 ohm resistor. BatteryAllowsContactorClosing = true

SSR+ and SSR- are shut,Led on, relays clicking inside battery.
Debug via USB shows battery data.
No error on Arduino IDE serial monitor.
Battery voltage shown by inverter.
Battery does not charge or discharge.
As the Plenticore settings allow any BYD battery as well as Pylontech, I have tried all CAN batteries.

dalathegreat, 2024-03-01:
Kostal inverters are listed as "⚠️ (Testers welcome!)", nobody has confirmed one to work yet.

Here are some interesting threads, BYD+Kostal: https://secondlifestorage.com/index.php?threads/communication-between-byd-battery-and-kostal-plenticore-inverter.10513/

Looks like it could also work via Modbus? 
[image]

Good luck with testing, not really anyone that has done this before you, so wishing for the best!

jspv33, 2024-03-02:
Ok, I was just hoping not to be the first one!
I will give it a try with Modbus. The inverter was up to now connected over Modbus to a BYD HV.
What I did find out with the CAN setup was when the power meter is disconnected from the inverter, the Leaf battery discharges into the inverter. Obviously there is nothing stopping the AC to be fed to the grid.

jspv33, 2024-03-05:
Testing with BYD MODBUS gives packet length error:

BMS Status: Active, , Power: 180W 
Values from battery
Real SOC%: 57% , GIDS: 149 (x77Wh) , Battery gen: 1 , Has heater: 0 , Max cell voltage: 3770mV , Min cell voltage: 3760mV , Cell deviation: 10mV [1;33m[E] 187664| ModbusServerRTU.cpp  [ 252] serve: RTU receive: E5 - Packet length error
[0m[1;33m[E] 188674| ModbusServerRTU.cpp  [ 252] serve: RTU receive: E5 - Packet length error
[0m[1;33m[E] 189684| ModbusServerRTU.cpp  [ 252] serve: RTU receive: E5 - Packet length error
[0m[1;33m[E] 190694| ModbusServerRTU.cpp  [ 252] serve: RTU receive: E5 - Packet length error
[0m[1;33m[E] 191704| ModbusServerRTU.cpp  [ 252] serve: RTU receive: E5 - Packet length error
[0mValues going to inverter

dalathegreat, 2024-03-05:
From the Fronius Wiki;

Make sure the Battery-Emulator has a good modbus connection to the Inverter

- Use shielded wires for a stable connection
- If you see lots of "ModbusServerRTU.cpp [ 252] serve: RTU receive: E5 - Packet length error" in the USB output on the LilyGo, it could be an indication that wiring is not perfect

dalathegreat, 2024-04-25:
@recomch Were you able to get the setup running?

dalathegreat, 2024-04-26:
Ok! Closing this issue for now then. Glad you got a working setup now! 🥳

closed 2024-04-26 by dalathegreat