dalathegreat/Battery-Emulator #222: Code quality - Adding Linting rules

gingofthesouth, 2024-03-11:
It would be nice to add some linting rules to this project and have a linter run as part of the merge request pipeline. If listing doesn't pass merge should be blocked until the pipeline steps no longer fail.

Just a thought.

lenvm, 2024-03-11:
Hi @gingofthesouth, we are running pre-commit on every pull request, which in turn runs clang format. Additionally unit tests are performed, and the code is compiled for all batteries, and all inverters. If any check fails, the pull request cannot be merged.

Which other linting rules do you have in mind to add to the pull request pipeline?

gingofthesouth, 2024-03-11:
> Hi @gingofthesouth, we are running pre-commit on every pull request, which in turn runs clang format. Additionally unit tests are performed, and the code is compiled for all batteries, and all [...]

Hi, Looks like I jumped the gun a bit here. Having used exclusively Gitlab and Jenkins pipelines I was looking in the wrong place for the pipeline info in the repo.

I'll look over the clang format config and get back to you if I have any suggestions but I have just run it locally and it seems to pick up the worst of my formatting mistakes.

closed 2024-03-15 by lenvm