dalathegreat/Battery-Emulator #225: CAN sending breaks when inverters change baud rate

dalathegreat, 2024-03-16:
### Problem description
Some inverters (like Sungrow) like to switch between 250kbps and 500kbps at startup, to see what type of battery is connected to it. This is a huge problem for us, since it can put the CAN bus into a faulted state that never recovers.

### Suggested solution
From RPI pelle_c

```
Believe I got this fixed now in rpi-version. Had to do a number of checks, and if one of them didn't pass the can-interface is restarted ("releasing notifiers", interface down/up), after 1-5 seconds (random).
Check that sending message returns ok
In addition, following is done on interval:
Check if the interface is reported as up
Check that the tx_counter on tne interface isn't the same as it was 60 secs ago
```

dalathegreat, 2024-03-17:
User @Newevg is piloting an attempted fix for this, two lines of code added to the `ESP32CAN.cpp` file;

```
ESP32Can.CANStop();
ESP32Can.CANInit(); 
```

[image]

Let's see if it helps anything long term!

dalathegreat, 2024-04-10:
@Newevg , did this help?

Newevg, 2024-04-10:
Good morning,Yes, I believe it helped as so far I haven’t had any issues Thank you 🙏 Sent from my iPhone

closed 2024-04-23 by dalathegreat