dalathegreat/Battery-Emulator #230: Solis single phase hybrid inverter possible?

corglass, 2024-03-22:
I see the team has managed to get this wonderful setup working on solis can based 3ph inverters.  

Would the same setup work for single phase solis inverters? Like this one

https://www.ginlong.com/uploads/file/Solis_Manual_RHI(3-6)K_5G_ENX_V1.1_nl.pdf

ffundoxx, 2024-03-28:
No, it won't for that one. This project is for high voltage batteries (EV batteries, 400V/800V), so any inverter that doesn't support HV batteries, will not work. You can check the battery support specifications for an inverter in its manual: in the one you linked you'll find "Battery voltage range: 42 - 58V", which means a low-voltage battery. If you can find a 1-phase inverter  model that supports HV batteries, then yes.

corglass, 2024-03-28:
Really appreciate you're time replying. Very helpful

calco88, 2024-04-08:
I will bet that this 1PH solis will work. Supports HV batteries. Currently saving up to get one and test if no one else beats me to it.

https://www.solisinverters.com/uploads/file/Solis_datasheet_S6-EH1P%283%2C8-11%2C4%29K-H-US_USA_V1%2C0_2024_01.pdf

Cabooman, 2024-04-10:
Closing as the original question is answered

closed 2024-04-10 by Cabooman