dalathegreat/Battery-Emulator #232: Solax X3 Ultra compatibility?

freddanastrom, 2024-03-25:
First of all, I've been following this project for a while. Great job with this!

I'm planning to replace my current non hybrid inverter with the new Solax X3 Ultra 25K.
X3 Ultra is specified to only be compatible with Solax batteries T-BAT H 5.8, T-BAT-SYS-HV-S 2.5 and 3.6

Would there be any chance that the current Solax code is emulating any of this batteries, so that the Ultra can use the same emulation?

If not. Would it be possible to emulate any of the compatible batteries, or are there something obvious that makes it impossible?
What can I do to support this development once I have the inverter in place?

StarkJohan, 2024-03-25:
We just don't know until someone has tested the Ultra and/or dumped some logs with compatible battery systems.

ilia92, 2024-05-03:
@freddanastrom have you bought the Inverter ? Did you start it ?

freddanastrom, 2024-05-03:
> @freddanastrom have you bought the Inverter ? Did you start it ?

Yes, have the inverter since a few weeks. But I'm still waiting for my electrician to install it, which will probably be a few more weeks until it can be done.
I also bought a small stack of batteries (T-BAT-SYS-HV-S36) for testing.

CAN communication is however new to me, never worked with it before.
So any guidance to how I can start logging the communication between the inverter and battery would be appreciated.

StarkJohan, 2024-05-07:
Do you happen to live in the southern parts of Sweden? 

The easiest way to get help with any parts of this project is joining the discord server. It's not free, but to me it's been essential in getting my system up and running. The time I've saved alone is worth the fee.

josiahhiggs, 2024-06-12:
> No, I live near Nyköping. so a bit of a distance to the southern parts.. I wasn't familiar with the memberships and Discord server,but will check it out and maybe give it a go.

I would highly recommend joining the Discord server! And the Emulator should work on the Ultra, Solax Triple Power LFP CAN.

DeadmaroZ-TLOTL, 2024-07-04:
With ultra inverter one problem, in main screen not show battery percentage, olways show 0%
[image]

[image]

[image]
[image]
!
[image]

[image]
[IMG_20240704_163827](https://github.com/dalathegreat/Battery-Emulator
[image]

[image]
/assets/150045808/bdcb0213-abf3-4022-ab34-51193a1cf730)
[image]

DeadmaroZ-TLOTL, 2024-07-04:
With ultra inverter one problem, in main screen not show battery percentage, olways show 0%
[image]

DeadmaroZ-TLOTL, 2024-07-04:
With ultra inverter one problem, in main screen not show battery percentage, olways show 0%

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

DeadmaroZ-TLOTL, 2024-07-05:
Yes, basically the only problem at the moment is the SOC (State of Charge) on the main screen!

Yes, I am using the dual Lilygo variant!

I didn't quite understand about the modbus?

Which CAN dump do you need?

StarkJohan, 2024-07-05:
Sounds very promising. 

I wonder if you are able to control the settings over modbus from e.g. Home Assistant? The existing integration should work but I'm just curious if it does. That way you can also force charge/discharge to check the battery function fully.

A CAN dump between the original battery and the inverter should give us all we need to get that last SOC sorted. The SOC likely just needs to go in an additional register for the Ultra compared to the inverters currently in use with the emulator.

DeadmaroZ-TLOTL, 2024-07-08:
Under these conditions, something doesn't quite work anymore? These pictures and videos are from the X3-Hybrid-G1/G2. Later, I'll also make some from the X3-Ultra.

https://photos.app.goo.gl/hDpPAeptM84SikHw9

[CAN_Log_X3-Hybrid-G1G2.txt](https://github.com/user-attachments/files/16123352/CAN_Log_X3-Hybrid-G1G2.txt)
[image]
[image]
[image]

DeadmaroZ-TLOTL, 2024-07-08:
[CAN_Log_X3-Ultra.txt](https://github.com/user-attachments/files/16123484/CAN_Log_X3-Ultra.txt)

DeadmaroZ-TLOTL, 2024-09-14:
Solax ultra log with Triple battery s7 https://f6340f99acf1.sn.mynetname.net:8099/index.php/s/fSeSw77aGCQ8Skr

Battery 2 is solax battery, battery 1 is tesla battery!

DeadmaroZ-TLOTL, 2024-11-19:
What else is needed to fully integrate the Solax Ultra?

dalathegreat, 2024-12-03:
Fix is now up as a PR, to get the SOC% to work in the app when using a Solax Ultra! https://github.com/dalathegreat/Battery-Emulator/pull/656

dalathegreat, 2024-12-28:
Closing this as completed. **Solax Ultra works perfectly** now with the addition of PR #704

closed 2024-12-28 by dalathegreat