dalathegreat/Battery-Emulator #234: Modbus eats too much resources, causes CAN to deadlock

dalathegreat, 2024-03-27:
### What
The Modbus protocol is taking too much CPU from time to time, causing timing of CAN messages to be incorrect. 

### Why is this an issue?
Worst case scenario a 10ms message can be delayed with almost 200ms, totally unacceptable on an embedded real-time device. BMW i3 opens contactors due to this.

### How to fix
We need to:
- Get Modbus to lower the CPU load
- Implement some warning signs of this
- More?

StarkJohan, 2024-03-27:
- Deactivate modbus entirely when only CAN is needed/used.

dalathegreat, 2024-03-29:
https://github.com/dalathegreat/Battery-Emulator/pull/236 is now merged, good first step in lowering CPU load!

dalathegreat, 2024-03-31:
Situation is much better with the latest v5.6.0 release, but could always be improved further!

dalathegreat, 2024-04-03:
Setting this one as resolved now, 5.7.dev is getting really fast! 🐎

closed 2024-04-03 by dalathegreat