dalathegreat/Battery-Emulator #238: Feature request: Add ability to save WiFi settings to flash

lenvm, 2024-03-27:
## What
Add ability to save the following WiFi settings to flash via the webserver.

https://github.com/dalathegreat/Battery-Emulator/blob/49fca8e0e065c27694fd25976e02bcd4365cb7c4/Software/USER_SETTINGS.cpp#L31-L32

This request is similar to the changes implemented in [PR #146]

## Why
- Implementing this allows the user to flash code without specific WiFi SSID and password to the board, connect to the WiFi access point of the board, and then add the WiFi SSID and password of the network they want the board to connect to. Potentially a reboot button would be needed in the webserver, to make the updated values take effect.
- Implementing this allows the user to update the board over the air with a software that is compiled without specific WiFi SSID and password. Flashing the binary that is compiled without specific WiFi SSID and password would still allow the board to connect to the WiFi network it was previously connected to. This helps the user to quickly update to the latest version of the software, without having to manually update the WiFi SSID and password in the source code.

Cabooman, 2024-04-10:
+1

Would be really useful

dalathegreat, 2024-07-09:
Added in PR https://github.com/dalathegreat/Battery-Emulator/pull/376

closed 2024-07-09 by dalathegreat