dalathegreat/Battery-Emulator #239: Feature request bucket: MQTT

Cabooman, 2024-03-29:
This issue tries to gather all MQTT related requests for future implementation. The current short list:

- BMS status included in common info topic, with discovery. Currently tricky because of the 'enum' device class, figure out how to do it. Maybe some info : https://community.home-assistant.io/t/wanting-to-map-sensor-number-value-to-words/40663
- LED color in common info as well, similar issue as BMS status
- Expand the command support by having the lilygo subscribe the command topics, and setting up the lilygo as a "commandable" device in discovery

GkoeTheMiller, 2024-03-29:
Useful commands would be:
"REBOOT" (reboot LilyGo)
"STOP" (execute open-contactor-sequence)
"START" (execute close-contactor-sequence - if no errors are present)
"ANALYSE" (do cell-analysis)
Commands to switch batteries (execute open-contactor-sequence, switch battery, execute close-contactor-sequence) or something like that...

eriksson25, 2024-05-19:
Could we change so that the MQTT uses the ssidAP name set in User_settings as identifier when publishing? The emulator webpage header now uses it (since PR312).

Sh0tgun98, 2024-11-10:
Control over Battery charge / discharge current limitation (LilyGo) would be very handly for automation.