dalathegreat/Battery-Emulator #242: Feature request : Add support for Volkswagen MEB BMS

nkrepo, 2024-04-03:
Hi all,
First to say that is a great project and congrats to all of you!
I would like to ask if there are plans to support Volkswagen MEB BMS ? I know probably is hard to get hold of a complete VW MEB battery and they are way more expensive than alternatives, but I assume in the feature they will be more readily available.

I noticed that there is work from @Tom-evnut  implementing the decoding of the BMS  from CAN [VW BMS](https://github.com/Tom-evnut/VW-bms). Is there any way to integrate it to Battery-Emulator ?

Tom-evnut, 2024-04-03:
This is the internal CAN bus of the battery not the external one you would need.

nkrepo, 2024-04-03:
Oh that's unfortunate, lets hope in the feature there will be documentation how to access the main BMS controller for the MEB batteries. Thank you for the quick reply !

nkrepo, 2024-04-09:
One question , related with the possibility of getting support for MEB platform BMS. The main BMS controller for MEB platform is on the battery pack or it is integrated with the rest of the car electronics ?

zeman51, 2024-04-09:
Hello. It is integrated into battery module.

dalathegreat, 2024-04-19:
See https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-battery-integration

mvgalen, 2024-12-04:
If I understand correctly, this issue was related to adding support for the complete MEB batteries. Initial support has been merged. So, I am closing this issue.

closed 2024-12-04 by mvgalen

nkrepo, 2024-12-05:
> If I understand correctly, this issue was related to adding support for the complete MEB batteries. Initial support has been merged. So, I am closing this issue.

Yes it is about complete MEB batteries, thank you for your contribution.