dalathegreat/Battery-Emulator #277: No contact between two Lillygo

OrustEngineering, 2024-04-30:
Hi

We set up two Lillygo to be used on Solax X3 G2/G3 inverter and a Nissan Leaf 2012 battery.
Currently we do not use any precharging since the inverter has this.

The battery Lilly comunicate with the battery but becomes red
[image]

The inverter Lillygo i green comunicate with the inverter which accept but get SOC valu set to 0% 

It seems it is the communication beetween the boards that do not work.

[image]

It proberly someting dead simple but have not found it.

Please help

Cabooman, 2024-04-30:
This is a known issue with more recent SW (5.7+), please downgrade to 5.6.0 until we've had the chance to find the root cause

dalathegreat, 2024-05-01:
Fixed in 5.9.0 , thanks for troubleshooting this @Cabooman ! 

@OrustEngineering , please update to 5.9.0 https://github.com/dalathegreat/Battery-Emulator/releases/tag/v5.9.0

closed 2024-05-01 by dalathegreat

Solarcs1979, 2024-05-03:
Bra jobbat!