dalathegreat/Battery-Emulator #280: BMW Hybrid Batteries

yngndrw, 2024-05-01:
Hello,

I know that support for the [BMW i3 batteries is mentioned](https://github.com/dalathegreat/Battery-Emulator/wiki/BMW-i3-battery) (With a caveat), but does anybody know if the HV hybrid batteries are supported? (For example the G20 330e battery)

Thank you!

dalathegreat, 2024-05-13:
Hi @yngndrw , yes the BMW 530E atleast has the same CAN messages as the BMW i3 battery!
https://github.com/damienmaguire/BMW-PHEV-BMS

[image]
This battery should be possible to use with the emulator. I am not sure if the 330E uses the same CAN. Someone just needs to test one :)

wjcloudy, 2025-01-19:
Beta support added in #804 

We still need can logs from an intact 330e/530e to finish it