dalathegreat/Battery-Emulator #286: SOFAR - LEAF bug

gkasprow, 2024-05-04:
I struggled to get the 30kW Leaf battery working and finally discovered that the RLY N relay signal was not activated.
So I connected RLY P with RLY N, and now it works. The precharge relay works as expected.
I'm not sure if I did it right.

dalathegreat, 2024-05-05:
If RLY P and RLY N are connected together, precharge wont happen (you will apply full voltage directly)

Are you using automatic contactor control via GPIO ? See this wiki for troubleshooting steps: https://github.com/dalathegreat/Battery-Emulator/wiki/Contactor-Control-via-GPIO-pins

dalathegreat, 2024-05-12:
The diagram under the LEAF battery was wrong, LV17 and LV11 was swapped. This has now been fixed: https://github.com/dalathegreat/Battery-Emulator/wiki/Nissan-LEAF-battery/6e2e63395cf06b23f2ad1fa5d728150aec6c2bac

Closing this issue as fixed!

closed 2024-05-12 by dalathegreat