dalathegreat/Battery-Emulator #288: proper HW

gkasprow, 2024-05-05:
I designed proper HW for emulator
https://github.com/gkasprow/battery_emulator/wiki
I'm waiting for comments from you, and I can add some features needed for other batteries.
I need them for LEAF batteries.

dalathegreat, 2024-05-05:
Nice work @gkasprow ! 

We are actually trying to move away from the LilyGo boards completely, due to the need for 2x CAN buses, and CAN-FD. This new board design that one Discord member is working on has built in relays for precharge and contactor control:

[image]

But having more alternatives is always better! Feel free to make a separate Wiki page about your add-on board!

gkasprow, 2024-05-05:
I like it! Do you have schematics? I'd like merge these two projects into one.

gkasprow, 2024-05-05:
We are working on open hardware projects for years. Example is https://sinara-hw.github.io/. It consits of nearly 90 sub-projects.
We have the sponsor for the boards. So I can produce let's say 10 pcs and give them away to the active community members for intensive tests.

LeighAS, 2024-05-05:
More peeps on the hardware can be no bad thing. Thanks Greg.

gkasprow, 2024-05-05:
Yes, but it's better to support and maintain a single HW version at least from a software point of view.
That's why I need schematics to make my HW compatible with the SW you are developing.

00martin00, 2024-05-13:
> I designed proper HW for emulator https://github.com/gkasprow/battery_emulator/wiki I'm waiting for comments from you, and I can add some features needed for other batteries. I need it for LEAF batteries. The schematics are here https://github.com/gkasprow/battery_emulator/files/15213206/Battery_emulator.PDF

be careful, the circuit diagramm has changed:
https://private-user-images.githubusercontent.com/81711263/328879150-d68caf7d-a0c3-4602-89bd-a701bb759d9a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2MjM5ODAsIm5iZiI6MTcxNTYyMzY4MCwicGF0aCI6Ii84MTcxMTI2My8zMjg4NzkxNTAtZDY4Y2FmN2QtYTBjMy00NjAyLTg5YmQtYTcwMWJiNzU5ZDlhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDE4MDgwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFlMjUzMTNmY2MwMjA5OTI4ZTIwNmI3YjlmZDg1MWQxNjQwYmQ1MGIxYTllMzY0ZjQzY2JjMTU5NWU0YWIwZWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.WIRpOhdJhME19GrJQ9YwGUeL4yIluKHOCX64Y0kfWkQ

gkasprow, 2024-05-25:
I added second CAN and fixed the updated schematic ; if there is interest I can also integrate Lilygo board into my board as a single design.
For the moment I ordered PCBs because I have the Lilygo boards and will use them.

nagyrobi, 2024-08-12:
@dalathegreat I see refrences in many places to Discord but don't see anywhere a link to join. Tried searching in Discord's "Discover" section but not found.

Alizainharoon, 2024-08-12:
This is a link to dala account in patreon . You can get the link from there
https://www.patreon.com/dala

في الاثنين، ١٢ أغسطس ٢٠٢٤, ١٢:٥١ م H. Árkosi Róbert <

Dutchlincoln, 2024-12-26:
> Nice work @gkasprow !
> 
> We are actually trying to move away from the LilyGo boards completely, due to the need for 2x CAN buses, and CAN-FD. This new board design that one Discord member is working on has built in relays for precharge and contactor control:
> 
> ![bild](https://private-user-images.githubusercontent.com/26695010/328025312-bca7242c-ca7b-4b8d-8eb4-c3e25e2b37ef.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyMDYzOTAsIm5iZiI6MTczNTIwNjA5MCwicGF0aCI6Ii8yNjY5NTAxMC8zMjgwMjUzMTItYmNhNzI0MmMtY2E3Yi00YjhkLThlYjQtYzNlMjVlMmIzN2VmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjI2VDA5NDEzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYzODgyMWFkOGExODQwNjkxOTUxYWE0ODljODUxZWM4OWNiM2VkMDRiNDJmMjViMDA1MThjZjE0YTlhM2VlMTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0EcLTNu2QgtOYhLgYai8dyKtbHWA73D2QeXTNO-0JmE)
> 
> But having more alternatives is always better! Feel free to make a separate Wiki page about your add-on board!

Is there any news about this board? I am very new here and ordered the lily boards and FD can converter, but i rather have it all on 1 PCB with less wires. As i need to start yet, i rather start with the latest versions. Is this board already available, and does it accept the already written software? i intend to use it on a BMW 83Kwh battery, which needs the FD can signal i discovered here. Very happy with all the great work done here by the way, keep up the good work!

gkasprow, 2024-12-26:
I plan to maka a new version with integrated Liligo components to improve reliability. I will make a batch of 5 so if U can wait I can make extra one. I usually use Aisler service.

nagyrobi, 2024-12-26:
+1 for me please, and let me know the cost

gkasprow, 2024-12-26:
First I need to test it in scenario with single 20kW Sofar inverter and 2 Leaf natteries 45kWh each. For the moment I use it with single battery. For dual battery I need separate CAN interfaces.

dalathegreat, 2025-01-02:
The code now supports 4x different Hardwares, LilyGo, Stark Board, 3LB, and ESP32 Devkit V1

```
/* Select hardware used for Battery-Emulator */
//#define HW_LILYGO
//#define HW_STARK
//#define HW_3LB
//#define HW_DEVKIT
```

If you want to make more hardware support, please make a Wiki page for your hardware, and submit a pull request with the HAL file.

I dont think this "issue" should be kept open, we are now regularly adding new hardwares into the mix as discussed on the Discord server.