dalathegreat/Battery-Emulator #300: Option in User_settings to set name

eriksson25, 2024-05-12:
Running several systems, and accessing them throe VPN to monitor them it would be awesome if we could set a system name that shows on webpage instead of "Battery Emulator" direct in the User settings file. It could have "Battery Emulator" as default and then it would be easy to change.

dalathegreat, 2024-05-19:
@eriksson25 
Added this in PR https://github.com/dalathegreat/Battery-Emulator/pull/312

Now the ssidAP name set in USER_SETTINGS.cpp will also be shown as header (Default is Battery Emulator)

[image]

eriksson25, 2024-05-19:
Awsome! Will test it out!

dalathegreat, 2024-05-25:
Closing, others approved the PR!

closed 2024-05-25 by dalathegreat