dalathegreat/Battery-Emulator #308: WIFI disconnects after random periods and never recovers

frasersimpson83, 2024-05-17:
Have been having issues for quite a while and have been trying all sorts of different options but still having issues.

Have discussed with other users who are also having the same issues, but a lot of users having no issues but that doesn't mean there isn't something that needs either fixing or work to allow it to recover itself.

Something as simple as having the WIFI module re-initialise itself and attempt a reconnect if nothing is connected for over 1 hour or something.   Surely it can't be that hard to have the WIFI retry if it's not seeing a connection.   I wouldn't think it would create too much load etc.   Even if it was every 2 hours or 4 hours etc, at least it would recover.    At the moment I'm having to go outside and get into my locked battery box to get to inverter, battery etc to reset things.    

Not a major hassle in itself, but feel it could be easily fixed with some recovery routines.

dalathegreat, 2024-05-18:
@frasersimpson83 ,
The Wifi logic has a 1second retry, and 30 second maximum retry. You could try raising these to double, and see if it helps?

https://github.com/dalathegreat/Battery-Emulator/blob/8905e4d74ad6e948e097c14a4bdc39cb6d11f614/Software/src/devboard/webserver/webserver.cpp#L33

That is the place. If your wifi is down for longer than 30 seconds, it could explain why it does not reconnect properly.

frasersimpson83, 2024-05-18:
Thanks Dala, I'll give it a go.  It's odd as I don't think my WIFI would be down for more than 30 seconds.   I work in the I.T industry and I use enterprise WIFI AP's and there's no reports of issues in the logging for the AP.   I have one at one end of my house and one at the other end of the house as it's long and thin to give me coverage.

It had been really reliable for the previous versions up until the early V5's and it seems to mostly have become an issue since the updates you did around the performance of the Lilygo......  it can happily run for 2-3 days and then that's it, won't reconnect until it's restarted.

I will extend the retry section out a bit and see if it affects things.........  thanks for the reply.

frasersimpson83, 2024-05-23:
Hi Dala,   I've had no disconnects for well over 4 days now where as before I would only get 3 days at the very most and usually 2 days, so maybe it's worth changing the default to 60 or 90 (I've used 90) instead of 30 seconds ??

dalathegreat, 2024-05-23:
Thanks for testing @frasersimpson83 , I opened a PR for this! Guess we can close this issue as it works well for you now :)